TheThingsNetwork/lorawan-stack #1125: "Cannot show events" in the Console for user admin

w4tsn, 2019-08-07:
#### Summary
When working with the Console to take a look into an Applications Event Stream (as admin) it first shows something for a few seconds and then enters "Cannot show events" error message, which then keeps on displaying even after browser refresh. The event stream does work in the CLI though.

#### Steps to Reproduce

1. Setup ttn-stack as of cli guide (with working device)
2. Setup the console client
3. Enter the console as admin and go to any application with a working device (check with CLI)
4. Open the "Data" view

[ttn-stack-config.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/3477276/ttn-stack-config.txt)

#### What do you see now?
I can subscribe to the events streams of applications, gateways and devices assigned to the user-id admin when logged in as admin over the CLI.

I can see the event stream of gateways in the Console.

I cannot see the event streams of applications and devices in the Console, although being admin and collaborator with `ALL` access rights.

#### What do you want to see instead?
Seeing the event streams of applications and devices for the owner admin in the console, when logged in as admin and having `ALL` access rights.

#### Environment

docker-compose. stack container 3.1.0. Public IP and DNS, behind ACME TLS. Only TLS ports exposed (8xxx and 1700, plus 1885->80 and 8885->443).

```
ttn-lw-stack version
The Things Network Stack for LoRaWAN: ttn-lw-stack
Version:             3.1.0
Go version:          go1.12.7
OS/Arch:             linux/amd64
```

#### How do you propose to implement this?
I have now idea right now, where the problem lies.

#### Can you do this yourself and submit a Pull Request?
I don't think I can do this right now.

bafonins, 2019-08-07:
Could check the console in the development tools of your browser for any errors/warnings?

w4tsn, 2019-08-08:
I checked the "Data" view with open dev tools, but there are no errors or warnings. I noticed though, since the device was offline first, I opened the view and only after 3 minutes I activated the device, that the view showed me "Event stream started" for about 3 minutes and as soon as data was coming in to the app the view disappears and shows the error "Cannot show events".

[image]

kschiffer, 2019-08-12:
Thank you very much for reporting this issue @w4tsn. 
This is caused by the JS SDK not accounting for the possibility of multiple event JSON objects arriving per stream chunk. Luckily this is easily fixed. I will follow up with a PR.

PR #1145 (2019-08-13): Fix parsing of stream chunks in JS SDK
#### Summary
Closes #1125

#### Changes
- Account for multi line stream chunks, with multiple separate json strings

#### Notes for Reviewers
Very likely the cause of #1125. I was able to reproduce and fix it with this diff.

#### Release Notes
- Fixed a bug that resulted in events streams crashing in the console.
closed 2019-08-13 by kschiffer