TheThingsNetwork/lorawan-stack #1131: Support specific JS-NS and JS-AS KEK labels and AS-ID

johanstokking, 2019-08-08:
#### Summary

Support specific JS-NS and JS-AS KEK labels and AS-ID

#### Why do we need this?

As we cannot (always) derive this from `application_server_address`

#### What is already there? What do you see now?

- KEK labels are now `{ns|as}:<{network|application}_server_address>`
- AS-ID is the host of `application_server_address`

#### What is missing? What do you want to see?

- Add `{network|application}_server_kek_label` to `EndDevice`
- Add `application_server_id` to `EndDevice`

#### How do you propose to implement this?

Add the aforementioned fields and use these fields if set, otherwise fallback to the current behavior of taking the host name for KEK label and AS-ID.

#### Can you do this yourself and submit a Pull Request?

Yes

bafonins, 2019-08-08:
what should be done to the console ui?

johanstokking, 2019-08-08:
There will be three new fields in `EndDevice`:

- `application_server_kek_label`
- `application_server_id`
- `network_server_kek_label`

These need to be edited with the JS component, also in the Console.

bafonins, 2019-08-08:
Could you file a new issue for this?

PR #1133 (2019-08-09): Use NS/AS KEKs and AS-ID if specified

#### Summary

Closes #1131 

#### Changes

- Use NS/AS KEKs if specified, fallback to host from address (current behavior)
- Use AS-ID if specified, fallback to host from address
- Small fix to actually wire `AppSKeyReq` and `HomeNSReq` to the JS interop server

#### Notes for Reviewers

- @htdvisser review at least API
- The 2 KB max length of the KEK label is to align with the root key ID and session keys ID; it may be a URL, reference a HSM key slot and parameter, etc
- The 100 bytes max length for AS-ID is arbitrary but should be enough

#### Release  [... cut]
closed 2019-08-09 by johanstokking