TheThingsNetwork/lorawan-stack #1134: Make NS/AS KEK labels and AS-ID editable

johanstokking, 2019-08-08:
#### Summary

Make NS/AS KEKs and AS-ID editable in the Console

#### Why do we need this?

User experience

#### What is already there? What do you see now?

No input fields

#### What is missing? What do you want to see?

Input fields. I would put this close to the component addresses; AS address + ID + KEK label and NS address + KEK label belong together.

Note that the KEK labels and AS-ID are only available in the JS.

The fields are;

- `application_server_kek_label`
- `application_server_id`
- `network_server_kek_label`

#### Can you do this yourself and submit a Pull Request?

No

@kschiffer please reassign if desired.

htdvisser, 2019-09-24:
Blocked by #579

johanstokking, 2019-11-27:
Mostly blocked by https://github.com/TheThingsNetwork/lorawan-stack/pull/1616

johanstokking, 2019-11-27:
Added home NetID and claim authentication code fields

johanstokking, 2019-12-10:
This is not entirely in progress by #1616; claim authentication code is not part of that (yet)

bafonins, 2020-02-25:
Regarding `claim_authentication_code` should the users be able to set the date and time, or only the date? @johanstokking

johanstokking, 2020-02-25:
@bafonins I think the date is fine, but internally it's a timestamp, so including time. If the user specifies 1-1-2021, it will be 1-1-2021 0:00:00, effectively 31-12-2020, so take care of that.

closed 2020-03-02 by bafonins