TheThingsNetwork/lorawan-stack #1146: Application view "link" page non-deterministic behaviour in console

furtiman, 2019-08-12:
#### Summary

If the application is linked to a network server using a created API key, the status of this linkage and user access rights for the application are displayed differently if the page if refreshed several times

#### Steps to Reproduce

If you are a member of `thethingsindustries` organisation at `demo.thethings.industries`, you can:
1. Log in to the console and go to `applications` -> `generic-node-test-ivan`
2. Go to the `Link` section from the sidebar
3. Refresh the page several times and observe different behaviour

If not - ~ask for organisation access (internal)~ create an application in the console at `demo.thethings.industries`, create an API key and [link the application to the network server](https://thethingsnetwork.github.io/lorawan-stack/v3.1.0/guides/getting-started/console/), then follow p 1-3 ^

#### What do you see now?

Same application, configured and linked in advance, two different views when refreshing several times:

[image]
[image]

#### What do you want to see instead?

Consistent display of correct information

#### Environment

Current demo cluster

Safari/Chrome

#### How do you propose to implement this?

I guess the first step is to figure out if the problem is in console or in as/ns

#### Can you do this yourself and submit a Pull Request?

cc @bafonins @kschiffer

furtiman, 2019-08-12:
I think I see one potential point of conflict:

I am both the creator of the app, and a member of `thethingsindustries`, and for this app these two entities have different access rights:

Me:
[image]

`thethingsindustries`:
[image]

Although in both cases I have the right to view and edit API keys, the rights differ in point `Link as Application to a Network Server...`

I assume these right might conflict  -> hence sometimes - the `insufficient rights` message in the second screenshot of the issue, and sometimes correct display

Thoughts? Does this sound like a possible scenario?
cc @johanstokking @kschiffer

kschiffer, 2019-08-13:
I was able to reproduce and it indeed seems like the backend has trouble with rights intersection. When I added the organization to my app (without linking rights), it was not possible to link the application anymore, although my user possessed sufficient rights for that.

The responses from the link endpoint alternate between
```json
{  
  "code":10,
  "message":"error:pkg/applicationserver:link (link failed)",
  "details":[  
    {  
      "@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace":"pkg/applicationserver",
      "name":"link",
      "message_format":"link failed",
      "correlation_id":"aeab3fdb0ffe49b48f21767a584b78af",
      "cause":{  
        "namespace":"pkg/auth/rights",
        "name":"insufficient_application_rights",
        "message_format":"insufficient rights for application `{uid}`",
        "attributes":{  
          "uid":"test-apppp"
        },
        "correlation_id":"41e3a62f4551460186a07b132e5cba13",
        "code":7
      },
      "code":10
    }
  ]
}
```
and
```json
{
  "code":7,
  "message":"error:pkg/auth/rights:insufficient_application_rights (insufficient rights for application `test-apppp`)",
  "details":[
    {
      "@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace":"pkg/auth/rights",
      "name":"insufficient_application_rights",
      "message_format":"insufficient rights for application `{uid}`",
      "attributes":{
        "uid":"test-apppp"
      },
      "correlation_id":"8caafa138f9945f8aaa3ecd2d427284a",
      "code":7
    }
  ]
}
```
with the problem persisting even after the organization has been removed as a collaborator.

So I think this is not a console issue, as the console only reacts to these backend errors (as expected). However, this to somewhat related #939.

johanstokking, 2019-08-13:
@furtiman please always assign someone and a milestone, otherwise it gets lost

furtiman, 2019-08-13:
Ok, I was not sure who to assign, will be more precise next time

htdvisser, 2019-08-15:
Just checked this with @furtiman on the latest `master` and the problem isn't there anymore, likely because of the membership refactor #1054 / #1107.

Since @kschiffer also indicated that he ran this on 3.1.0, I'm going to close this issue.

closed 2019-08-15 by htdvisser