TheThingsNetwork/lorawan-stack #1147: Unable to update password as admin

johanstokking, 2019-08-13:
#### Summary

Unable to update password as admin through CLI

#### Steps to Reproduce

1. `$ ttn-lw-cli user update karleric --password <secret>`
2. `$ ttn-lw-cli user update karleric --temporary-password <secret>`

#### What do you see now?

Forbidden path errors, like:

```
error:pkg/rpcmiddleware/validator:field_mask_paths (forbidden path(s) in field mask)
    forbidden_paths=[password]
    correlation_id=de1536c5b64a4059802a65986b2a4e4c
```

#### What do you want to see instead?

Successful password update

#### Environment

`v3.1.0`

#### How do you propose to implement this?

Allow `password` and `temporary_password` for admins

#### Can you do this yourself and submit a Pull Request?

Can review

ZeroSum24, 2020-07-09:
@htdvisser it appears this might still be an issue? I've attempted to reset passwords for both admin and general users (working with v3.8.4) as an admin and was unsuccessful in all attempts including attempting to reset the admin password of the admin account I was using.

> ttn-lw-cli users update admin --password Example-password-1
```
error:pkg/rpcmiddleware/validator:field_mask_paths (forbidden path(s) in field mask)
    forbidden_paths=[password]
    correlation_id=9b56dc608aab477195b3762aa1eaac7c
```

If you would be able to help with this that would be much appreciated!

johanstokking, 2020-07-10:
Looks like we need some documentation here, and potentially change the CLI.

Please use `--temporary-password` instead.

ZeroSum24, 2020-07-10:
@johanstokking thanks very much for getting back to me!

I am not sure what the >  --temporary-password field should do. My assumption is that it either replaces the users current password with a temporary one, or provides a temporary password that the user may use in conjunction with their original.

What I have found on experimentation is that the >  --temporary-password field currently does neither of these things. I am not able to sign into the account using either the password I specify using the flag, nor the response in the temporary_password field. This is the same regardless of whether I set the > --require-password-update field. I am still able to sign into the account using the users original password though (although it required a few attempts with > --require-password-update field active).

> ttn-lw-cli users update general-user --temporary-password ljlksjfiomv03
> {
>   "ids": {
>     "user_id": "general-user"
>   },
>   "created_at": "2020-07-07T12:33:50.595Z",
>   "updated_at": "2020-07-10T13:58:59.104Z",
>   "temporary_password": "PBKDF2$sha256$20000$9aN9xG-1QaEOrLUl$e_ELaLNzY0l3m0OUO1M1bn1JeD_JucjK1WavmMBxQ-E",
>   "temporary_password_created_at": "2020-07-10T13:58:59.099Z",
>   "temporary_password_expires_at": "2020-07-12T01:58:59.099Z"
> }

I hope these extra details help with the issue.

johanstokking, 2020-07-10:
OK, we definitely need to clear up documentation here. If setting a temporary password, that would indeed need to take precedence, so that you can reset the password as admin.

So your use case is specifically to set the password for your users, right?

Or are the passwords forgotten? In that case you can use the `forgot-password` command.

ZeroSum24, 2020-07-13:
@johanstokking that's our use case aye. Thanks for looking into this.

PR #1192 (2019-08-22): Allow admins to set temporary user password

#### Summary

Closes #1147 if @johanstokking is okay with only allowing setting a temporary password (and not the actual password).

#### Changes

- Changed IS to allow admins to set a temporary password for users
	- Question: Should this trigger an email to the user?
- Some cleanup

#### Notes for Reviewers

I think it's bad practice for admins to be able to set real user passwords, at least until we can force them to change the password (#1191).

The API change doesn't affect external compatibility, it just removes the confusing zero `password_upd [... cut]
PR #3095 (2020-08-19): Improve error reporting when trying to change password with CLI

#### Summary

This improves error reporting in the CLI when trying to update the password of a user using the wrong command. Updating the password of a user is only allowed with the `update-password` command. It is however allowed for admins to set a temporary password on users, as well as an expiration time.

Closes #1147

#### Notes for Reviewers

This wasn't documented, so didn't add or update documentation.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: Th [... cut]
closed 2020-08-19 by htdvisser