TheThingsNetwork/lorawan-stack #1149: log of mac commands

blackenstock, 2019-08-13:
#### Summary
When investigating on ADR and timesync (on different platform) it is needed to have the mac commands logged of a node. And also have them forwarded in SWAGGER and the HTTP Integration and MQTT(make sense?)

I’ve looked into Swager integration, http integration with paste bin, mqtt but none seem to provide this information. How can I most easily find mac command history

Regards

Michael

#### Why do we need this?
to get an idea was or why the node is not acting as intended:
* to debug the timesync feature
* to improve the development of the endnode, to see if it gets an ADR response

#### What is already there? What do you see now?
normal payload will be provided

#### What is missing? What do you want to see?
* file export of the traffic including mac command
* integrate the mac command in the forwarding REST call to SWAGGER and HTTP integration

#### Environment
WebUI / SWAGGER / HTTP Integration / MQTT

#### How do you propose to implement this?

#### Can you do this yourself and submit a Pull Request?
no
https://www.thethingsnetwork.org/forum/t/log-of-mac-commands/28438

johanstokking, 2019-08-13:
Thanks @blackenstock for reporting. I think this is very useful indeed. We won't make this available on the application layer, i.e. MQTT, webhooks and AS gRPC, since this is very much network layer.

@rvolosatovs can we make this as events available, including the payload, for both directions?

blackenstock, 2019-08-19:
michael.aebli@gwf.ch, lukas.kempf@gwf.ch

rvolosatovs, 2019-08-19:
Network Server publishes events whenever it receives a MAC command request/response and whenever it schedules a MAC request/response, including payload(if present)
E.g. https://github.com/TheThingsNetwork/lorawan-stack/blob/42b63f81dfbd5a46580f31db5b96358f47196659/pkg/networkserver/mac_link_adr.go#L27-L29
@blackenstock can you clarify what is missing here?

rvolosatovs, 2019-08-27:
@blackenstock any update?
Can you confirm that the existing event infrastructure fulfills your requirement?

rvolosatovs, 2019-09-24:
Closing due to lack of input

closed 2019-09-24 by rvolosatovs