TheThingsNetwork/lorawan-stack #1158: SensorThings Webhook format

adriansmares, 2019-08-13:
#### Summary

We should add support for the [SensorThings Observation](http://docs.opengeospatial.org/is/15-078r6/15-078r6.html#31) format for the webhooks.

#### Why do we need this?

In order to support the [SensorThings Sensing API](http://docs.opengeospatial.org/is/15-078r6/15-078r6.html).

#### What is already there? What do you see now?

Only the JSON and Protobuf webhook formats.

#### What is missing? What do you want to see?

A new [format](https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/applicationserver/io/web/format.go) that should derive an SensorThings Observation based on the decoded payload.

#### Environment

9aeee5050dff3ec121e9b2656bcfd619ab0e3ac5

#### How do you propose to implement this?

I believe the format should strongly resemble the JSON format, but instead derive an observation for each of the provided fields inside the decoded payload. The Datastream IOT ID should be derived from the `EndDeviceIdentifiers.DeviceID` present inside the uplink provided to `FromUp`.

#### Can you do this yourself and submit a Pull Request?

Yes, but I will leave this to the community.

mihirr93, 2020-02-18:
@adriansmares I am grateful to you for your consistent support and guidance. It would be of great help if you could provide a similar guide describing the required steps like the last time. However, I had following doubts before we proceed and I would like to clarify the same:

1) Sensor Things API requires the sensor observation to be posted to the server URL (for eg. [http://iot.gis.bgu.tum.de:8081/FROST-Server-gi3/v1.0/Observations](http://iot.gis.bgu.tum.de:8081/FROST-Server-gi3/v1.0/Observations)) in the following format:
```
var Message =  { payload: { "result": humidity_sensorreading, "Datastream": {"@iot.id": 1}} };
Message.headers = {"Content-type" : "application/json"}
return Message;
```
As it can be seen, there are 3 major components. URL where it needs to be posted, sensorvalue, and the iot.id of the datastream of that particular sensor. Now, from what I understood, we are trying to use existing http webhook for implementing this. So the URL will be stored in the webhook integration. The payload value will be derived from the decoded payload field. But I am wondering, how the formatter will obtain the @iot.ids required for this?  You mentioned that it will be derived from the `EndDeviceIdentifiers.DeviceID`, but is it something which user can modify? And, there are multiple iot.ids to be provided by the user (one for each sensor). Can this particular field store multiple values? 

I imagine it as an empty field in the dashboard where user can enter the number of sensors in the payload and specify the correspoding @iot.id's for each of them.

2) In the last comment of the issue #1148 you mentioned the payload formatter twice: In the step 3 and the step 5. 

> Let me try to explain the whole pipeline better, maybe this will shed some light on how this works
> 
> 1. A device sends 1 uplink.
> 2. The 1 uplink is received by the gateway and sent to the Gateway Server which forwards it to the Network Server, which then forwards it to the Application Server. The message that reaches the AS is of type *ttnpb.ApplicationUplink.
> 3. The AS takes this 1 uplink and attempts to decode its binary payload into the decoded fields (turn the bytes in the FRMPayload field into the decoded structure of DecodedPayload) using the **device payload formatter**.
> 4. It then wraps it up in a *ttnpb.ApplicationUp and sends it to the frontends (MQTT, webhooks, PubSub, application packages).
> 5. The frontends receive this 1 *ttnpb.ApplicationUp and based on their settings (the format) choose which **formatter** to use (the formatter interface is defined in pkg/applicationserver/io/formatters).
> 6. The frontend calls formatter.FromUp with the 1 *ttnph.ApplicationUp and receives a slice of byte ([]byte), representing the body of 1 message.

Are both of these formatters same or different? I understand the first formatter is for decoding/encoding depending on the type of payload (custom or cayenne) where as the second formatter is formatting the data into the publishing format. Is it correct? Where and how would our formatter fit in?

[I aim to complete this task before the end of this month, do you think it is feasible?]

Thank you once again for your time.

htdvisser, 2021-03-26:
This has been inactive for quite some time. Is there still demand for this, or should we drop it?

adriansmares, 2021-03-26:
The demand for this format is very low and in general we no longer  want custom formats to be supported via webhooks. Packages may pickup external services instead.

closed 2021-03-26 by adriansmares