TheThingsNetwork/lorawan-stack #1160: Deleting a user over CLI does not remove the database entries

w4tsn, 2019-08-14:
#### Summary
When calling `ttn-lw-cli users delete someadmin` the user gets deleted as of the cli commands respons, login over Console and CLI is not possible anymore and `users get` does yield in `user othermo not found`. However when trying to re-create said user the cli responds with either `entity already exists` with mail address as cause or  when using another mail it complains that the uid of the user is already present.

#### Steps to Reproduce

1. setup ttn-lw-cli as of getting started
2. create a user (tested with `--admin`)
3. delete the user
4. try re-creating the same user

#### What do you see now?
**In the CLI** I cannot create a user with same uid and mail that I previously deleted.

**In the database** after deletion of a user all data in `accounts`, `users`, `user_sessions` etc. is still present.

#### What do you want to see instead?
Being able to recreate a user I deleted with the same uid and mail address, since the uid is used as login name.

#### Environment

```bash
The Things Network Stack for LoRaWAN: ttn-lw-stack
Version:             3.1.0
Go version:          go1.12.7
OS/Arch:             linux/amd64
```

#### How do you propose to implement this?
If this has something to do with deactivation/deprecation of a user, provide a separate function for complete deletion. Otherwise make the delete operation fully delete users.

It might be also good to also delete things like open `user_sessions`, but I'm not sure about the implications of this.

#### Can you do this yourself and submit a Pull Request?
No

htdvisser, 2019-08-14:
This is currently working as intended. Deletes are "soft", so entities are just marked as deleted, and they will stay in the database as such until we have either GC or recovery of deleted entities: #604. 

Also related to #1120 (for email address) and #1121.

closed 2019-08-14 by htdvisser

w4tsn, 2019-08-21:
Would it then make sense to have a force option or a command to remove "soft"-removed entities? That way I could remove something if I intend to, but not accidentally. Of course this would not protect me like a real recovery feature of a false intention removal that I'd like to revert later on.