TheThingsNetwork/lorawan-stack #1173: Add more context to events

htdvisser, 2019-08-14:
#### Summary

We could add some more context to our events.

#### Why do we need this?

Just because we have it, it may also help users when debugging.

#### What is already there? What do you see now?

We have support for adding "contextual" information to events, but aren't using it.

#### What is missing? What do you want to see?

- IP address of the RPC caller
- User agent of the RPC caller
- ID of the auth token that was used on the RPC

#### How do you propose to implement this?

I already looked into this some time ago, and considered adding it as details to the `Event` API. This would work well for RPC events, but would be awkward for uplink/downlink events so I didn't do that. Here's what I already had:

- Use `peer.FromContext`, and check the `x-forwarded-for` headers in the metadata

```go
var remoteAddrs []string
if md, ok := metadata.FromIncomingContext(ctx); ok {
  if forwardedForHeaders, ok := md["x-forwarded-for"]; ok {
    for _, forwardedForHeader := range forwardedForHeaders {
      for _, remoteAddr := range strings.Split(forwardedForHeader, ",") {
        remoteAddrs = append(remoteAddrs, strings.TrimSpace(remoteAddr))
      }
    }
  }
}
if peer, ok := peer.FromContext(ctx); ok {
  peerAddr := peer.Addr.String()
  if peerAddr != "pipe" { // Skip the gRPC loopback conn.
    if host, _, err := net.SplitHostPort(peerAddr); err == nil {
      peerAddr = host
    }
  }
  remoteAddrs = append(remoteAddrs, peerAddr)
}
if len(remoteAddrs) > 0 {
  details.CallerAddress = strings.Join(remoteAddrs, ", ")
}
```

- Check the metadata for a `user-agent` or `grpcgateway-user-agent`

```go
if md, ok := metadata.FromIncomingContext(ctx); ok {
  if userAgents, ok := md["grpcgateway-user-agent"]; ok { // Set by gRPC gateway.
    details.CallerUserAgent = strings.Join(userAgents, " ")
  } else if userAgents, ok := md["user-agent"]; ok {
    details.CallerUserAgent = strings.Join(userAgents, " ")
  }
}
```

- Check the metadata for a `bearer` token, extract the type and ID (but never the secret).

```go
if md := rpcmetadata.FromIncomingContext(ctx); strings.ToLower(md.AuthType) == "bearer" {
  if tokenType, tokenID, _, err := auth.SplitToken(token); err == nil {
    details.TokenType, details.TokenID = tokenType, tokenID
  }
}
```

#### Can you do this yourself and submit a Pull Request?

Yes, but I don't really have time for it now.

htdvisser, 2020-07-14:
@neoaggelos please check what's still TODO here

neoaggelos, 2020-08-10:
The only thing that still remains here is the `User-Agent` field. After a bit of testing, this is the behaviour of the two relevant headers:

- `grpcgateway-user-agent`: Set to the HTTP/gRPC client that is sending the request (e.g. `curl/7.68.0`, ` Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.105 Safari/537.36`, `ttn-lw-cli go/1.14.6 ttn/3.9.0 (f2316b5, 2020-08-06T14:44:26Z) grpc-go/1.29.1`)
- `user-agent`: Always overwritten and set to the stack version (e.g. `ttn-lw-stack go/1.14.6 ttn/3.9.0-dev grpc-go/1.29.1`)

The `grpcgateway-user-agent` seems more relevant, so will give priority to that one when both headers exist.

PR #3072 (2020-08-14): Add user agent metadata to published events

#### Summary

Closes #1173 

#### Changes

- Add `UserAgent` field to `rpcmetadata.MD`
- Add `UserAgent` field to published events when that is available in the context.

#### Testing

Unit testing, test locally by editing a device using curl, the cli, the console and verifying the user agent on the emitted events.

##### Regressions

There should not be any

#### Notes for Reviewers

Added as a new option, included by default.

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The c [... cut]
closed 2020-08-14 by neoaggelos