TheThingsNetwork/lorawan-stack #1175: Remove @bind decorator from class components

bafonins, 2019-08-14:
#### Summary

We should remove `@bind` decorators from class components

#### Why do we need this?

https://github.com/andreypopp/autobind-decorator#discouraged-approaches

#### What is already there? What do you see now?

A lot of class components annotated with `@bind`

#### What is missing? What do you want to see?

Only certain methods that need access to `this` being decorated

#### Environment

browser (oauth, console)

#### How do you propose to implement this?

1. Only bind methods that need access to `this`
2. Use arrow functions
```
class Component extends React.Component {
  ...
  handleClick = () => {}
  ...
```

The second approach might require some changes to `eslint` configuration

#### Can you do this yourself and submit a Pull Request?

Yes, but help is welcome.

kschiffer, 2019-08-14:
Mildly related:
https://github.com/TheThingsNetwork/lorawan-stack/blob/master/config/eslintrc.yaml#L24-L26

Decorating just before export statements is also discouraged since babel 7. We should take that into account as well if we do a refactor.

kschiffer, 2019-11-25:
> 1. Only bind methods that need access to `this`
> 2. Use arrow functions
```
class Component extends React.Component {
  ...
  handleClick = () => {}
  ...
```

Let's agree on using 1. only for consistency, since we already refactored some components to do this.

closed 2020-05-12 by bafonins