TheThingsNetwork/lorawan-stack #1199: Add linting checks to githooks

kschiffer, 2019-08-19:
#### Summary
Since our move to mage (#11) we have no more linting checks in our githooks. These checks helped us avoiding linting issues before we opened PRs. We should reintroduce these checks to our githooks.

#### Why do we need this?
Avoid linting issues that can prolong PR reviews. It also caused us to amass a couple of linting warnings over time, as we did not see these immediately: #1169 

#### What is already there? What do you see now?
Mage targets for linting (`go:lint`, `js:lint`, `styl:lint`)

#### What is missing? What do you want to see?
- Logic to lint only staged files
- Adding lint checks to githooks

#### Environment
Universal.

#### How do you propose to implement this?
Basically fiddling with our mage setup and adding the appropriate targets to `.mage/git.go`

#### Can you do this yourself and submit a Pull Request?
Yes, but help is very much welcome.

kschiffer, 2019-09-24:
After the discussion today, about the issues with local go linting, I would propose to add linting pre-push hooks only for the webui `js:lint`, `styl:lint`, which usually run through quickly.

htdvisser, 2020-08-24:
Good enough to have CI fail on this.

closed 2020-08-24 by htdvisser