TheThingsNetwork/lorawan-stack #1200: Add feature flags to the console

kschiffer, 2019-08-19:
#### Summary
The console should provide certain functionality only if the current stack setup (enabled components, current rights) would allow supporting this functionality.

#### Why do we need this?
Based on the configuration and right setup, we already know which functionality would not work. It makes sense to not display this functionality in the console, as we would otherwise wrongfully suggest that it is still possible to use.

#### What is already there? What do you see now?
The SDK respecting the `stackConfig` (including what components are enabled) and avoids request to components that are known to be disabled.

#### What is missing? What do you want to see?
A way to easily disable components (and functionality in general) based on the known list of rights and disabled components.

#### Environment
Console.

#### How do you propose to implement this?
This appears like a very good use case for a higher-order component. I could envision a usage like this:
```js
@withRequirement(['as', 'is'], ['RIGHT_APPLICATION_LINK', 'RIGHT_APPLICATION_INFO'])
class MyComponent extends React.PureComponent {
…
}
```
Additionally, for a more fine-grained control inside rendering functions, we should add a wrapper component to check a set of requirements against the current context and configuration. It would then only render its children when the requirement passes:
```js
render () {
  …
  <Requires components=['as', 'is'] rights=['RIGHT_APPLICATION_LINK', 'RIGHT_APPLICATION_INFO']>
    <Link to='…'>Link</Link>
  </Requires>
  …
}
```

Both utilities would get the rights from the store (needs yet to be added there actually) and the component data from the global environment.

#### Can you do this yourself and submit a Pull Request?
Yes. 

Input welcome cc @bafonins

bafonins, 2019-08-28:
What about the `<SideNavigation />` entries?

kschiffer, 2019-09-05:
So I think we need to split this into three aspects:
1. Hiding links from the navigation (top, side and tab navigation)
2. Preventing access to views with disabled features (should be mostly a redirect to the parent view)
3. Hiding bits inside views that are not accessible (eg. the device table inside the application overview, when the user has no right to view devices)

All in all, I think it would be good to have a central module that determines whether a feature is available or not, which we can then refer to for all three cases. This enables us to change conditions for particular features easily.

```js
// pkg/webui/console/lib/feature-toggles.js

export mayViewApplicationDevices = () => 
  userHasRights([RIGHT_APPLICATION_DEVICES_READ])
  && stackComponentsEnabled(['is']) // I know, checking for 'is' here is unnecessary, it's just for illustration purposes
…
```

For hiding links in the navigation, we could simply compose the navigation entries object conditionally:
```js
…
    entries: [
      {
        title: sharedMessages.overview,
        path: matchedUrl,
        icon: 'overview',
      },
      ...(mayViewApplicationDevices()
        ? [
            {
              title: sharedMessages.devices,
              path: `${matchedUrl}/devices`,
              icon: 'devices',
              exact: false,
            },
          ]
        : []),
…
```
Not particularly slick, but there are likely libraries that enable use to compose this less verbose.

For preventing access, I see two possible ways:
1. Use a HOC to decorate the views, checking the accessibility (using the module mentioned above) and redirect if need be
2. Use basically the same HOC but wrap it around at the `<Route />` level

Don't see any particular grave advantage or disadvantage for both options. Do you?

For hiding parts of a render, I see two options again:
1. Use a wrapper component, that gets fed with the condition (from the module above) and only renders when it is met
2. Use basic conditional rendering, like we already do a lot for other concerns. (like `mayViewApplicationDevices() && <Component />`)

The wrapper might give us the opportunity to easily fine-tune the behavior later on.

> `withRequirement` -> `withAuthorization`. what do you think?

Do you mean to simply rename, or merge with `with-auth.js`? Might be an idea, since both are authorization related. If we use option 2 above, I will look into whether that can be done reasonably. We currently use `WithAuth` as a wrapper around the whole `<Switch />` tree. So if we combine both, it would need to be more fine-grained. We earlier had an `<AuthRoute>` wrapper component, to be used inside `<Switch />`. This would allow us to choose distinctive conditions for each route.

bafonins, 2019-09-16:
> For hiding links in the navigation, we could simply compose the navigation entries object conditionally

I think it would easier to refactor side navigation and main navigation using the compound components approach

> For hiding parts of a render, I see two options again:
> Use a wrapper component, that gets fed with the condition (from the module above) and only renders when it is met

👍

> Do you mean to simply rename, or merge with with-auth.js?

I mean just renaming. It has nothing to do with authentication.

PR #1543 (2019-11-07): Add feature toggles to the console
#### Summary
Closes #1200 

This PR will add feature toggles to the console. All views that the user is not allowed to view will automatically redirect back. Navigation links will not be displayed and specific components inside views will be hidden if the feature requirements are not met.

#### Changes
- Add user rights to redux store
- Add `<Require />` component for conditionally rendering children based on feature availability
- Add `withFeatureRequirement()` HOC that basically wraps the `<Require />` component but also adds simplified redirection
- Add `f [... cut]
closed 2019-11-07 by kschiffer