TheThingsNetwork/lorawan-stack #1204: Fix rights modification of collaborators with more rights than current user

kschiffer, 2019-08-20:
#### Summary
It is currently not possible to properly manage the rights of a collaborator that has more rights than oneself. The problem here is that if User A has rights that are outside of the scope of granted rights for User B, User B cannot modify the rights of User A properly because:
- Sending the out-of-scope rights together with the legitimately altered rights will result in a 403 Forbidden
- Not sending the out-of-scope together with the legitimately altered rights will not result in a 403 but will revoke these rights, even when that might not be desired

#### Steps to Reproduce

- Call the `SetCollaborator` RPC on an entity that has more rights than the current user
- See a 403 if you pass the out-of-scope rights in the `rights` prop (legitimately attempting to leave them enabled)
- See a 200 if you do not pass them, but the rights are now revoked

#### What do you see now?
Inability to set rights of collaborators correctly if they have more rights than oneself.

#### What do you want to see instead?
Being able to selectively revoke rights that the current user does not have without having to revoke them altogether. Likewise, being able to leave these rights enabled while modifying other rights.

#### Environment
Irrelevant.

#### How do you propose to implement this?

#### Can you do this yourself and submit a Pull Request?
No. Assigned @htdvisser but feel free to reassign to whoever is more appropriate.

See also #939 and https://github.com/TheThingsNetwork/lorawan-stack/issues/939#issuecomment-522966655

htdvisser, 2019-08-20:
The implementation of this would look something like below for `pkg/identityserver/*_access.go`

```diff
@@ -202,12 +203,26 @@ func (is *IdentityServer) setGatewayCollaborator(ctx context.Context, req *ttnpb
 	if err := rights.RequireGateway(ctx, req.GatewayIdentifiers, ttnpb.RIGHT_GATEWAY_SETTINGS_COLLABORATORS); err != nil {
 		return nil, err
 	}
-	// Require that caller has at least the rights we're giving to the collaborator.
-	if err := rights.RequireGateway(ctx, req.GatewayIdentifiers, req.Collaborator.Rights...); err != nil {
-		return nil, err
-	}
+	newRights := ttnpb.RightsFrom(req.Collaborator.Rights...)
 	err := is.withDatabase(ctx, func(db *gorm.DB) error {
-		return is.getMembershipStore(ctx, db).SetMember(
+		store := is.getMembershipStore(ctx, db)
+		existingRights, err := store.GetMember(
+			ctx,
+			&req.Collaborator.OrganizationOrUserIdentifiers,
+			req.GatewayIdentifiers,
+		)
+		if err != nil && !errors.IsNotFound(err) {
+			return err
+		}
+		// Require the caller to have all added rights.
+		if err := rights.RequireGateway(ctx, req.GatewayIdentifiers, newRights.Sub(existingRights).GetRights()...); err != nil {
+			return err
+		}
+		// Require the caller to have all removed rights.
+		if err := rights.RequireGateway(ctx, req.GatewayIdentifiers, existingRights.Sub(newRights).GetRights()...); err != nil {
+			return err
+		}
+		return store.SetMember(
 			ctx,
 			&req.Collaborator.OrganizationOrUserIdentifiers,
 			req.GatewayIdentifiers,
```

bafonins, 2019-09-06:
@htdvisser Here is your solution adjusted to make sure that any user with the `RIGHT_APPLICATION_SETTINGS_COLLABORATORS` right can delete another collaborator even if the latter has more rights. Please confirm we want to allow that. See the diff:

```diff
// application_access.go
-    // Require that caller has at least the rights we're giving to the collaborator.
-    if err := rights.RequireApplication(ctx, req.ApplicationIdentifiers, req.Collaborator.Rights...); err != nil {
-        return nil, err
-    }
 
    err := is.withDatabase(ctx, func(db *gorm.DB) error {
-        return is.getMembershipStore(ctx, db).SetMember(
+        store := is.getMembershipStore(ctx, db)
      
+        if len(req.Collaborator.Rights) > 0 {
+            newRights := ttnpb.RightsFrom(req.Collaborator.Rights...)
+            existingRights, err := store.GetMember(
+                ctx,
+                &req.Collaborator.OrganizationOrUserIdentifiers,
+                req.ApplicationIdentifiers,
+            )
 
+            if err != nil && !errors.IsNotFound(err) {
+                return err
+            }
+            // Require the caller to have all added rights.
+            if err := rights.RequireApplication(ctx, req.ApplicationIdentifiers, newRights.Sub(existingRights).GetRights()...); err != nil {
+                return err
+            }
+            // Require the caller to have all removed rights.
+            if err := rights.RequireApplication(ctx, req.ApplicationIdentifiers, existingRights.Sub(newRights).GetRights()...); err != nil {
+                return err
+            }
+        }
 
+        return store.SetMember(
            ctx,
            &req.Collaborator.OrganizationOrUserIdentifiers,
            req.ApplicationIdentifiers,
```

bafonins, 2019-09-06:
Besides that, I also noticed that our current implementation of the api keys/collaborators form does not reflect the actual meaning of entity rights enforced in the is.

As @htdvisser noted, `RIGHT_{entity}_ALL` rights do not only mean **all current rights**, but also **all future rights**. So, atm it is not possible to assign all entity rights as a list of `RIGHT_{entity}_{whatever, except ALL}`. If the user checks all checkboxes or uses the `select all` checkbox, the console would always sends the `RIGHT_{entity}_ALL` right which is not always desirable. One might want to give a user/organization just the current rights on an entity, but the future rights. This should be reflected in the console and could look like this:
[image]
[image]

Let me know what you think

kschiffer, 2019-09-06:
Indeed. This would also simplify the implementation a lot, as we could then get rid of the translation of universal rights into all individual rights, and all the ambiguity surrounding this.

PR #1345 (2019-09-18): Fix rights modification of collaborators with more rights than…

#### Summary

Closes #1204 

#### Changes

- Change implementation of `SetApplicationCollaboratorRequest`
- Change implementation of `SetCollaboratorCollaboratorRequest`
- Change implementation of `SetGatewayCollaboratorRequest`
- Add tests for each entity to test new handling of collaborator rights

#### Notes for Reviewers 

~~@htdvisser I left the clients access rpc without changes, since it seems to be inherently different from applications, gateways and collaborators.~~

@johanstokking @rvolosatovs no need for review
@kschif [... cut]
closed 2019-09-18 by bafonins