TheThingsNetwork/lorawan-stack #1205: Console support for organizations

bafonins, 2019-08-20:
#### Summary

The console should provide functionality to manage organizations:
* List organizations
* Create new organizations
* Update existing organizations
* Delete organizations

#### Why do we need this?

The users should be able to manage organizations through the console

#### What is already there? What do you see now?

ui components
example layouts/functionality could be taken from applications/gateways

#### What is missing? What do you want to see?

* Entry in the main navigation (similar to applications/gateways)
* `Organizations` service in the js sdk
* side navigation entries for Organization sub-views:
  * Overview
  * Collaborators
  * API keys
  * General Settings
  * anything else
* API entries in the console api definitions `console/api`
* Redux setup for organizations
See also:
organization collaborators: https://github.com/TheThingsNetwork/lorawan-stack/issues/27
organization api keys: https://github.com/TheThingsNetwork/lorawan-stack/issues/251

#### Environment

browser (console)

#### How do you propose to implement this?

Similar to applications/gateways

#### Can you do this yourself and submit a Pull Request?

yes, but feel free to take over

cc @kschiffer @htdvisser

htdvisser, 2019-08-20:
You could indeed copy-paste existing stuff from applications/gateways with a couple of important differences:

1. Only users can collaborate on organizations (not organizations)
2. You also need to be able to set APPLICATION/CLIENT/GATEWAY rights on API Keys and Collaborators (not only ORGANIZATION rights)

And it may also be good to keep in mind that it is possible (with the CLI, it should be possible with the console) to create Applications/Clients/Gateways with the initial collaborator being an Organization (and not the user).

bafonins, 2019-08-21:
> And it may also be good to keep in mind that it is possible (with the CLI, it should be possible with the console) to create Applications/Clients/Gateways with the initial collaborator being an Organization (and not the user).

@kschiffer could you think how we can implement this in the ui?

kschiffer, 2019-08-21:
I think doing it in the create forms is better:

For all actions that can be performed as a user <b>and</b> organization, we add a dropdown somewhere at the top of the form to indicate as what entity the action should be performed.
This is the approach is quite common and used eg. .by github or docker hub. 
<details><summary>see examples</summary> [image][image]
</details>
We also need to make sure that it is clear to the user whether he/she is regarding an entity as organization or plain user.

The organization index page would be a list of organizations (as said, similar to gateways/apps).
For single organizations, it would be a sidebar view consisting of:
- List of collaborators (as index page, no need for an overview here)
- API Keys
- General Settings (including contact info)

So it would be a pretty simple view/sub-view to start with, though I am sure there will be some peculiarities unearthed while we work on it.

htdvisser, 2019-08-22:
Let's not worry about creating applications/gateways/... with an organization as initial collaborator just yet. For now creating them under the user works perfectly fine. The user can just make the organization a collaborator.

> We also need to make sure that it is clear to the user whether he/she is regarding an entity as organization or plain user. 

Apart from creation, there is no difference.

> List of collaborators (as index page, no need for an overview here)

For consistency I think it would still be good to have an overview. This overview would be similar to the User overview and could show a list of Applications/Gateways where the Organization is collaborator on, similar to how the Application Overview has a list of End Devices.

bafonins, 2019-09-17:
Will continue after https://github.com/TheThingsNetwork/lorawan-stack/pull/1345

closed 2019-10-21 by bafonins