TheThingsNetwork/lorawan-stack #1216: Console support for users

htdvisser, 2019-08-22:
#### Summary

The console should provide functionality to manage the current user.

#### Why do we need this?

Users should be able to manage their general information and API keys.

#### What is already there? What do you see now?

We have existing API and support in the CLI. Very similar to Applications/Gateways and Organizations (#1205).

#### What is missing? What do you want to see?

- Link in the top-right dropdown
- Anything needed in `sdk/js`
- Anything needed in `console/api`
- Navigation similar to Organizations (#1205)
	- Overview
	- API Keys (#251)
	- General Settings

#### Environment

Browser

#### How do you propose to implement this?

Similar to #1205

bafonins, 2019-08-22:
cc @kschiffer

kschiffer, 2020-08-11:
Profile settings would be a part of the account app. #1422 

For API Keys, I could for now see a simple menu entry in the profile dropdown, pointing to a view with a table of all configured API keys by the user, with an option to add new user scoped API key.

kschiffer, 2021-06-01:
Profile settings and User API Keys can now be managed in the Console or Account App. As such I think this can be closed. @htdvisser please reopen if you think otherwise.

htdvisser, 2021-06-01:
I think it's okay for now, but eventually we'll probably have to extend the functionality of user management that's available to admins. What's currently still missing is:

- Attributes (but that's also still missing from Organizations)
- Contact Info (but I think that's also still missing from all other entities)
- Generating/Setting a temporary password
- Setting the state description (for example: the reason why someone was rejected)
- Setting/clearing the profile picture (for example: to remove offensive profile pictures)
- Managing API keys
- Managing user sessions

KrishnaIyer, 2024-10-22:
This is all completed

closed 2024-10-22 by KrishnaIyer