TheThingsNetwork/lorawan-stack #1218: OTAA join fails

berrySkies, 2019-08-22:
#### Summary

The OTAA join request from my device fails. It appears that the join-accept message received by the app does not get forwarded to the gateway.

#### Steps to Reproduce

1. Use an OTAA join request from device. (My device is running software from LMIC MCCI library on TTGO module)
2. The join request is visible on the MQQT server and console
3. Look to see if join-accept is received by gateway
[stackConfig.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/3529510/stackConfig.txt)

#### What do you see now?

[image]
[image]

...

#### What do you want to see instead?

...

#### Environment

The Things Network Command-line Interface: ttn-lw-cli
Version:             3.1.0
Go version:          go1.12.7
OS/Arch:             linux/amd64

Using MacOS and Docker
...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

berrySkies, 2019-08-24:
I'm going to close the issue. Following further investigations I've now managed to get downlink messages sent to gateway and wireless concentrator.

closed 2019-08-24 by berrySkies

Lucianovici, 2020-02-13:
Hi, can you please share with us how you fix the issue, I have a similar behaviour with a Multitech Conduit gateway. Please check: https://github.com/TheThingsNetwork/lorawan-stack/issues/1988

johanstokking, 2020-02-13:
May be closed by #1989