TheThingsNetwork/lorawan-stack #1223: "Special" password character set is non-extensive and opaque for the user

rvolosatovs, 2019-08-22:
#### Summary

"Special" password character set is non-extensive and, worst of all, opaque for the user.

https://github.com/TheThingsNetwork/lorawan-stack/blob/7b6050dba6ca1bb546f90570c23402cb059ca0ba/pkg/identityserver/user_registry.go#L84

#### Steps to Reproduce

1. Go to https://join.thethings.industries/
2. Attempt to create an account with password `mtMp`2mhbtytvsN?)`Zf(W/]`

#### What do you see now?

`Should contain at least one special character` error and a major confusion

#### What do you want to see instead?

Successful creation of account or at least error message stating which characters stack expects to be present in the password as "special".

#### Environment

Any

#### How do you propose to implement this?

Either the "special" password character set should be extended, or the "special" character set should be defined clearly, such that users know, which characters we consider "special".

#### Can you do this yourself and submit a Pull Request?

yes

htdvisser, 2019-08-23:
As I commented on Slack yesterday:

Let's redefine that "special char" category to:

- Any char smaller than `0`
- Any char between `9` and `A`
- Any char between `Z` and `a`
- Any char larger than `z`

johanstokking, 2019-08-23:
> * Any char smaller than `0`

If we start doing numeric char comparison, then I'd suggest "Any char between ` ` (space) and `0`. 

Chars under `0x20` are non-printable and space itself is quite confusing as a "special char", but we could allow it as such.

PR #1239 (2019-08-26): Broaden the IS password character limits, allow Unicode

#### Summary

Closes #1223 

#### Changes

- Broaden IS password character limits
- Support Unicode

#### Notes for Reviewers

It's a very small change, but I assigned both @johanstokking and @htdvisser to take look, since it's quite important that we agree on this.
Using unicode characters should not be a problem - it's 2019 already. This also gives a lot of freedom to our users, which I'm sure (at least) some will be happy about.
closed 2019-08-26 by rvolosatovs