TheThingsNetwork/lorawan-stack #1225: Gateway activity update and wrong time rendering

johanstokking, 2019-08-22:
#### Summary

Gateway activity update is not updated and last active time rendering is wrong.

#### Steps to Reproduce

1. Go to one of the connected gateways in the demo environment, i.e. https://demo.thethings.industries/console/gateways/647fdafffe000162
2. See last activity time

#### What do you see now?

These issues are probably separate:

1. The time is shown as `-X`, where `X` is likely seconds. This seems regression; it worked before
2. Sometimes `X` stalls
3. Sometimes, when a gateway connects, events are shown correctly, but it keeps shown as **Disconnected**

[image]

#### What do you want to see instead?

1. The old behavior; i.e. `X seconds ago` 
2. Always keep incrementing `X`, or switch correctly to **Disconnected** state
3. Keep the **Connected** and **Disconnected** states updated

#### Environment

Latest `master` deployed currently at https://demo.thethings.industries

#### How do you propose to implement this?

Not sure

#### Can you do this yourself and submit a Pull Request?

Can review

bafonins, 2019-08-23:
The issue seems to appear only when using Safari.

KrishnaIyer, 2019-08-23:
@bafonins: I noticed this yesterday on Chrome as well.

bafonins, 2019-08-29:
> Sometimes, when a gateway connects, events are shown correctly, but it keeps shown as Disconnected

moved to https://github.com/TheThingsNetwork/lorawan-stack/issues/1242

closed 2019-08-29 by bafonins