TheThingsNetwork/lorawan-stack #1226: client 'console' not found error when logging in

swattor, 2019-08-22:
#### Summary

When logging into the web console, the following error is produced:

Not Found
client `console` not found

The "take me back" button returns you to the /oauth URL indicating that the login successfully completed

...

#### Steps to Reproduce

1. Spin up stack
2. Configure
3. Attempt to login 

[ttn_config.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/3530616/ttn_config.txt)

#### What do you see now?

...

#### What do you want to see instead?

...

#### Environment

...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2019-08-22:
> 1. Spin up stack
> 2. Configure

Can you elaborate a bit on the second point?

Did you follow the Getting Started, especially the part where you need to [create an OAuth client for the Console](https://thethingsnetwork.github.io/lorawan-stack/v3.1.0/guides/getting-started/running-the-stack/)?

swattor, 2019-08-22:
Ah. 

See, the documentation is a bit confusing:

https://github.com/TheThingsNetwork/lorawan-stack/blob/master/doc/gettingstarted.md links to:
https://ttn.fyi/v3/getting-started which links to:
https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.0.4/doc/gettingstarted.md

Which doesn't have the bit about the console.

I've now run the create OAuth client for the console, and now I get 'Internal Server Error'!

swattor, 2019-08-22:
It's worth noting that I'm running this on a seperate machine (so accessing via http://remote-ip/ rather than http://localhost). I'm not sure whether that is relevant.

swattor, 2019-08-22:
Tried various things, including starting again. While following the instructions gets me further, I'm now stuck with an "Internal Server Error" when I login.

htdvisser, 2019-08-23:
Updated the redirect [ttn.fyi/v3/getting-started](https://ttn.fyi/v3/getting-started) to point to the latest getting started guide.

htdvisser, 2019-08-23:
@swattor Can you please expand the **Steps to Reproduce** section with the **exact** steps you took? Without detailed steps we can't reproduce the issue. The issue template also asks you to upload your configuration, terminal output / logs and screenshots (you can redact sensitive information). Please do that so that we can investigate this further.

htdvisser, 2019-08-23:
Seems like you're not the only one running into a 500 error, so marking this as a possible duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/1232.

htdvisser, 2019-08-23:
Hi @swattor, If you're able to compile and run The Things Stack from source, try it with the code in the Pull Request https://github.com/TheThingsNetwork/lorawan-stack/pull/1236

swattor, 2019-08-23:
Yeah, I had tracked it down to something to do with oauth.

Looking at the logs provided in the duplicate, it looks very similar, or at least failing in the same way. I will try and run with debug and provide some useful info in that ticket.

fernandobarreto, 2019-09-06:
Hi @htdvisser , I tried the last release of docker ttn stack (3.1.2) and the "Internal Server Error" still remains at console page.

johanstokking, 2019-09-06:
Can you enable debug logs on the server (`TTN_LW_LOG_LEVEL=debug` in Docker compose environment for the stack container) and copy server logs?

Also, can you as @htdvisser requested before, copy your **exact** steps and update the original comment in this issue?

fernandobarreto, 2019-09-09:
I find out the problem, it was on certificates.... Thanks anyway.

johanstokking, 2019-09-09:
Sure, you're welcome!

closed 2019-09-09 by johanstokking

GMouaad, 2020-01-17:
> I find out the problem, it was on certificates.... Thanks anyway.

Any Explanation how you resolved the issue please ? 
I'm running the Stack on a VM, so i assumed it has smth to do with the redirects or csrf ??
I get the following error:
`{"code":16,"message":"error:pkg/errors/web:unknown (Unauthorized)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/errors/web","message_format":"Unauthorized","attributes":{"message":"No auth cookie"},"code":16}]}
`