TheThingsNetwork/lorawan-stack #1230: Single Host Configuration

htdvisser, 2019-08-23:
#### Summary

I think it would be a good idea to add a `--common-deployment` (or similar) flag to `ttn-lw-stack` and `ttn-lw-cli` from which the other address/URL flags can be derived.

#### Why do we need this?

Because right now it's a bit too much config.

#### What is already there? What do you see now?

If you want to configure `ttn-lw-stack` for a different host than localhost, you have to change all of:

```
TTN_LW_CONSOLE_OAUTH_AUTHORIZE_URL="http://localhost:1885/oauth/authorize"
TTN_LW_CONSOLE_OAUTH_TOKEN_URL="http://localhost:1885/oauth/token"
TTN_LW_CONSOLE_UI_AS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_CONSOLE_UI_CANONICAL_URL="http://localhost:1885/console"
TTN_LW_CONSOLE_UI_GS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_CONSOLE_UI_IS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_CONSOLE_UI_JS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_CONSOLE_UI_NS_BASE_URL="http://localhost:1885/api/v3"
TTN_LW_IS_EMAIL_NETWORK_CONSOLE_URL="http://localhost:1885/console"
TTN_LW_IS_EMAIL_NETWORK_IDENTITY_SERVER_URL="http://localhost:1885/oauth"
TTN_LW_IS_OAUTH_UI_CANONICAL_URL="http://localhost:1885/oauth"
TTN_LW_IS_OAUTH_UI_IS_BASE_URL="http://localhost:1885/api/v3"
```

Similarly, for `ttn-lw-cli`:

```
TTN_LW_APPLICATION_SERVER_GRPC_ADDRESS="localhost:8884"
TTN_LW_DEVICE_TEMPLATE_CONVERTER_GRPC_ADDRESS="localhost:8884"
TTN_LW_GATEWAY_SERVER_GRPC_ADDRESS="localhost:8884"
TTN_LW_IDENTITY_SERVER_GRPC_ADDRESS="localhost:8884"
TTN_LW_JOIN_SERVER_GRPC_ADDRESS="localhost:8884"
TTN_LW_NETWORK_SERVER_GRPC_ADDRESS="localhost:8884"
TTN_LW_OAUTH_SERVER_ADDRESS="https://localhost:8885/oauth"
```

#### What is missing? What do you want to see?

Since many deployments will likely use a "common" pattern (the one we'll provide in the Getting Started), we might as well provide a flag that reduces repetition. A `--common-deployment=demo.thethingsstack.io` would configure all above settings to use the `https://` scheme, the `demo.thethingsstack.io` host and the default ports for these "common" deployments (so HTTPS on 443, gRPC on 8884).

#### How do you propose to implement this?

Add logic such as: "if `common-deployment` flag set, then for each of the above flags, set it to the common format"

#### Can you do this yourself and submit a Pull Request?

I can, if this is something we want.

htdvisser, 2019-08-23:
An alternative idea would be to have a `ttn-lw-stack quickstart` or `ttn-lw-cli quickstart` wizard that would generate a config file based on some detected things and some questions to users. I'll work that idea out next week.

johanstokking, 2019-08-26:
> An alternative idea would be to have a `ttn-lw-stack quickstart` or `ttn-lw-cli quickstart` wizard that would generate a config file based on some detected things and some questions to users. I'll work that idea out next week.

This is very complementary and also very handy, especially if we can print out a yaml and env file.

It's not entirely alternative as having a public FQDN setting also affects future config that contains a host name, and it makes configuration easy, regardless of generating it.

htdvisser, 2020-07-14:
@neoaggelos please see what's still TODO in here.

neoaggelos, 2020-08-24:
For this, I think we should just go with the `common-deployment` flag suggestion. The `quickstart` command sounds useful, but would not really work well with the rest of the Getting Started guide. Also, it might cause issues in the future with new configs.

Keeping this on Next Up, because I am currently busy with higher priority issues, namely #1406.

danrs, 2020-12-03:
It would be good to mention this in the [configuration instructions](https://thethingsstack.io/getting-started/installation/configuration/), specifically the section at the bottom discussing deployments on localhost using the machine's static IP in place of `localhost`.

benolayinka, 2020-12-07:
> An alternative idea would be to have a `ttn-lw-stack quickstart` or `ttn-lw-cli quickstart` wizard that would generate a config file based on some detected things and some questions to users. I'll work that idea out next week.

@neoaggelos if possible, i definitely vote for `quickstart`. Prompts are great for making steps digestible. One `common-deployment` option which optionally overwrites other options feels a little bit less intuitive.