TheThingsNetwork/lorawan-stack #1234: Create devices without keys

johanstokking, 2019-08-23:
#### Summary

Create devices without keys

#### Why do we need this?

For devices whose keys are provisioned on an external JS

#### What is already there? What do you see now?

Currently we can disable the JS component and not enter the OTAA information

#### What is missing? What do you want to see?

Even if the JS component is enabled in the cluster, we should still be able to not enter keys, implicitly requiring a JS client flow

#### How do you propose to implement this?

When creating devices, explicitly allow for not entering or generating keys. The `root_keys` should really be `nil`. If the join-request still ends up on the cluster-local JS, it will simply return an error that there are no keys.

#### Can you do this yourself and submit a Pull Request?

Can review, but let's discuss how we do this first

rvolosatovs, 2019-08-26:
In CLI you can just specify `--join-server-enabled=false` regardless of whether JS is enabled in the cluster or not.
For console I suppose we could just have a checkbox, e.g. "external join-server", which, when checked, would disallow specifying root keys and not make requests to JS registry.

johanstokking, 2019-08-26:
Indeed CLI has this already builtin. Clarified OP.

htdvisser, 2019-08-27:
Move discussion to https://github.com/TheThingsNetwork/lorawan-stack/issues/579

johanstokking, 2019-09-19:
We pushed https://github.com/TheThingsNetwork/lorawan-stack/issues/579 to next up, but we need a quick and easy way not to enter root keys in the Console in the short term.

My suggestion would simply be a checkbox (default checked when JS is there) when creating a device, that shows or enables the root key fields.

htdvisser, 2019-09-19:
Should we perhaps reword this issue to "hide device key fields when not applicable"? Then we could also hide the NwkKey, SNwkSIntKey and NwkSEncKey fields for pre-1.1 devices, as this is a bit confusing now.

bafonins, 2019-09-19:
To clarify:
1. This is only applicable for OTAA devices
2. If JS is in the cluster `root_keys.app_key.key` and `root_keys.nwk_key.key` fields are disabled by default.

Something like this:
[image]

Is this correct?

rvolosatovs, 2019-09-19:
1. Yes
2. No. The purpose of this is to be able to create a device on all components except JS, regardless of whether or not JS is present in the cluster. Console should provide an alternative to CLI flag `--join-server-enabled=false`. This is possible to do implicitly via root keys, but IMO the simplest and cleanest way to do this is what I proposed in https://github.com/TheThingsNetwork/lorawan-stack/issues/1234#issuecomment-524826626

The proposal looks fine, however I would rename `Disable keys` to something like `External Join Server`, as I mentioned in https://github.com/TheThingsNetwork/lorawan-stack/issues/1234#issuecomment-524826626. If we have that we can also, for example, require that `Join Server Address` is set to something we can use(in the future).
In fact, this checkbox should be checked when the device is ABP and there should be no way to change it's value(`Join Server Address` field should also be disabled).

bafonins, 2019-09-20:
> I would rename Disable keys to something like External Join Server

👌

> In fact, this checkbox should be checked when the device is ABP

Currently, if the device is ABP nothing is sent to js and all js related values are removed from the payload. What is more, I dont see any reasons to show or keep the `External Join Server` checkbox checked since its value is wont be included in the submitted payload, see the form for ABP devices:
[image]

> If we have that we can also, for example, require that Join Server Address is set to something we can use(in the future)
> Join Server Address and Resets Join Nonces fields should also be disabled

So should the `Join Server Address` field be disabled or not?

Updated create view:
[image]

johanstokking, 2019-09-20:
> Currently, if the device is ABP nothing is sent to js and all js related values are removed from the payload. What is more, I dont see any reasons to show or keep the `External Join Server` checkbox checked since its value is wont be included in the submitted payload

That's good

> So should the `Join Server Address` field be disabled or not?

Yes

> Updated create view:
> [image]

Yes, this is good

johanstokking, 2019-09-25:
@bafonins everything in #1374 works, but I have one minor request;

If your go from external JS to cluster JS, you need to fill out the JS address yourself. You can leave it empty, but then nothing happens. Can we prefill this with the JS address?

bafonins, 2019-09-25:
> Can we prefill this with the JS address?

The js address of the current cluster?

johanstokking, 2019-09-25:
Yes!

closed 2019-09-30 by bafonins