TheThingsNetwork/lorawan-stack #1235: C-style array copy spacing and braces

johanstokking, 2019-08-23:
#### Summary

C-style array copy spacing and copy braces improvements

#### Steps to Reproduce

1. Select the `< >` button near EUIs or keys
2. Click copy

#### What do you see now?

The clipboard contains, for example: ` 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08`

1. Notice the space at the start (and not at the end)
2. Notice no `{` and `}`

#### What do you want to see instead?

1. No spacing
2. Preferably adding braces

#### How do you propose to implement this?

No idea, is probably quite simple

#### Can you do this yourself and submit a Pull Request?

Can review

@kschiffer please adjust milestone or assignment

htdvisser, 2019-08-23:
Not having the curly braces is intentional. Not all languages use `{}` for arrays; some use `[]` instead.

bafonins, 2019-08-23:
we removed braces here: https://github.com/TheThingsNetwork/lorawan-stack/pull/155

kschiffer, 2019-08-28:
So what should we prefer: braces or no braces @johanstokking ?

johanstokking, 2019-08-28:
OK let's leave the braces and then only fix the first space that gets copied. Mind the labels and the milestone; this is no priority at all

johanstokking, 2019-08-28:
Updated OP

PR #1259 (2019-09-03): Remove leading space from safe inspector copy value
#### Summary
Closes #1235 
References #1086 

#### Changes
- Remove leading whitespace in the copied value of the safe inspector (when in c-style mode)
- Fix proptypes

#### Notes for Reviewers
Just putting this in-between as it was an easy fix…

cc @johanstokking
closed 2019-09-03 by kschiffer