TheThingsNetwork/lorawan-stack #1240: Add ttn-lw-cli configuration endpoint

johanstokking, 2019-08-26:
#### Summary

Add endpoint in The Things Stack for serving the `ttn-lw-cli` config file

#### Why do we need this?

To download the concerning configuration file, most importantly including the host names of the gRPC services and potentially CA to use

#### What is already there? What do you see now?

Manual configuration

#### What is missing? What do you want to see?

A public endpoint to download the `ttn-lw-cli` configuration.

Bonus would be;
- Add a command in the CLI that takes the host name and downloads the `yml` file
- By default saves it to `ttn-lw-cli.yml` in the working dir (confirm overwrite file etc)
- Uses `ttn-lw-cli.yml` from the working dir (before checking `$HOME/.ttn-lw-cli.yml`) if present

That way, users can simply:

```
$ ttn-lw-cli use join.thethings.industries
$ ttn-lw-cli login
```

#### How do you propose to implement this?

For serving the configuration, simply a public endpoint to `GET` the configuration. I guess that YAML is all we need, but we may consider other formats through `Accept`

#### Can you do this yourself and submit a Pull Request?

Can review

htdvisser, 2019-09-10:
We'll simplify this to generate a config in the CLI (without calling the server) for a standard deployment on `host`.

johanstokking, 2019-10-02:
Insight from #1408;

- With `--local` (default), put the dotfile in the working dir (`os.Getwd()`)
- With `--user`, put the dotfile in the user config dir (`os.UserConfigDir()`)
- Extend `pkg/rpcmiddleware/discover` to return the endpoints based on the user provided hostname. Probably introduce a new struct there with the discovery result. If #138 isn't closed yet, return sane defaults (i.e. all components enabled and use current default port fallback)

neoaggelos, 2020-03-27:
With #2150 merged, is there anything else that we would like to have for this, or can we close the issue?

johanstokking, 2020-03-27:
In #2150 I suggest to use `pkg/rpcmiddleware/discover` instead of hardcoding the default ports in the CLI.

Please extract https://github.com/TheThingsNetwork/lorawan-stack/blob/feature/pba/pkg/rpcmiddleware/discover/discover.go#L60-L63 to an exposed function and use that to get the address for TLS/non-TLS based on user input instead.

johanstokking, 2020-03-27:
Changing labels as it's not broken but can be nicer. Since it works, lowering priority.

PR #2299 (2020-04-15): Improve cli config file generation

#### Summary

Closes #1240

#### Changes

- Use service discovery for the gRPC service ports
- Add `--grpc-port` and `--oauth-server-address` options in use command.

#### Notes for Reviewers

...

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [X] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [X] Documen [... cut]
closed 2020-04-15 by neoaggelos