TheThingsNetwork/lorawan-stack #1242: Improve gateway status UI in console

bafonins, 2019-08-26:
#### Summary

Currently the Gateway Overview page has two status indicators:
[image]
[image]
[image]

Such UI is quite confusing.

#### Why do we need this?

Current gateway status UI is ambiguous.

#### What is already there? What do you see now?

1. `<GatewayStats />` component that displays the **status of a gateway depending on the response from `Gs.GetGatewayConnectionStats`**
2. `<EventsWidget/>` component that displays the **events connection status**
3. `<GatewayOverview />` component that combines both

#### What is missing? What do you want to see?

Unambiguous and clear gateway connection status display.

#### Environment

browser (console)

#### How do you propose to implement this?

UI changes in the gateway overview

#### Can you do this yourself and submit a Pull Request?

Yes, but we have to discuss the solution first.

cc @kschiffer @johanstokking

johanstokking, 2019-08-26:
Good point to improve this. It's super useful as it is, but it's a bit confusing indeed.

I think we need to show a better separation between gateway activity and events. For people looking at a gateway, activity typically means that it's connected and that there's traffic going back and forth. That is were "real time" makes most sense, including the "X seconds ago", not really the events (which are rendered timestamped anyway).

So my suggestion would be:

- Remove "X seconds ago" from events, keep them only for gateway stats
- Drive the stats by the events
   - on `gs.up.receive`; increment up and bump last seen
   - on `gs.down.send`; increment down and bump last seen
   - on `gs.status.receive`; bump last seen
- Increment stats periodically in the background in case we miss events. Here a longer interval works, like 10 secs

> * Remove status indicator from the events widget (?)

I would show something that events are "connected", but not the "seconds ago" thing

> * Dont show events for disconnected gateways (?)

We can't really do this as connected/disconnected depends on GS being in the cluster

> * Add tooltips to both status indicators (?)

Yes

kschiffer, 2019-08-28:
I agree with @johanstokking. We don't need the last-seen indicator for events, but rather for the connection status only.

This is also somewhat connected to #770, where I also wanted to look at cleaner ways to display this information. Will hopefully be able to work on this soon.

closed 2019-09-02 by bafonins