TheThingsNetwork/lorawan-stack #1256: Add password complexity validation in oauth app

bafonins, 2019-08-28:
#### Summary

We should add password complexity validation in the oauth app.
For more details see https://github.com/TheThingsNetwork/lorawan-stack/pull/1244 and https://github.com/TheThingsNetwork/lorawan-stack/pull/1239

#### Why do we need this?

Instant feedback to the user if the password is not valid.

#### What is already there? What do you see now?

* The Create Account view
* The Update Password view
* Length/match validation

#### What is missing? What do you want to see?

Password complexity validation (check for digits, uppercase chars and special chars)

#### Environment

browser (oauth)

#### How do you propose to implement this?

1. Hardcode unicode characters, see [this app](https://apps.timwhitlock.info/js/regex)
2. Adapt some 3rd party solution

#### Can you do this yourself and submit a Pull Request?

Yes, but feel free to reassign

bafonins, 2021-02-19:
@kschiffer isnt still already implemented?

kschiffer, 2021-02-19:
Yeah it actually is. Was just skimming this and though this is about adding a "password-strength-o-meter" sort of thing.

closed 2021-02-19 by kschiffer