TheThingsNetwork/lorawan-stack #1260: Issue with installation of The Things Network Stack on Raspberry Pi Model 3B

KuroTetsu25, 2019-08-29:
**Summary**
Unsuccessfull Installation of The Things Network Stack on Raspberry Pi Model 3B

**Steps to Reproduce**
Follow the installation based on the official documentation on this github repository

**What is already there? What do you see now?**
docker-compose run --rm stack is-db init
Starting lorawan-stack_310_linux_armv7_redis_1    ... done                                                                                                                  Starting lorawan-stack_310_linux_armv7_postgres_1 ... done                                                                                                                  standard_init_linux.go:190: exec user process caused "exec format error"

**What is missing? What do you want to see?**
It should running successfully

**Environment**
Raspberry Pi Model 3B with armv7l architecture

johanstokking, 2019-08-29:
@KuroTetsu25 thanks for reporting.

The Docker image is built for `amd64` architecture, hence it doesn't work on `arm7l`. We have an open issue about this; https://github.com/TheThingsNetwork/lorawan-stack/issues/510, therefore I'll mark this issue as duplicate and close it. You may upvote the aforementioned issue to put it higher on the radar. Note it's currently blocked by limitations in our delivery pipeline.

For the time being, and I think this would be my advised approach for RPi anyway, is to use `systemd` and don't use a container approach to avoid the overhead.

johanstokking, 2019-08-29:
Duplicate of #510

closed 2019-08-29 by johanstokking

KuroTetsu25, 2019-08-29:
Thank you for your reply, Johan