TheThingsNetwork/lorawan-stack #1282: Automatic linking on application create

johanstokking, 2019-09-02:
#### Summary

Automatic linking on application create

#### Why do we need this?

To avoid the hassle of creating an API key and setting the link

#### What is already there? What do you see now?

Manual creation of API key and setting the link

#### What is missing? What do you want to see?

A checkbox on application create to link the application.

This is only supported if there's a AS in the cluster

#### How do you propose to implement this?

On successful create in the Entity Registry and if the checkbox is checked, create a new API key with the `RIGHT_APPLICATION_LINK` right and probably a name to indicate it's created by the Console/CLI and set the link on AS.

#### Can you do this yourself and submit a Pull Request?

Can review

htdvisser, 2019-09-02:
As I already said in the offline discussion, the idea was to turn the "Create Application" flow into a wizard, similar to what we do with the "Create End Device".

Probably doing it with a checkbox saves some time, but makes the process too much "magic" IMO.

kschiffer, 2019-09-02:
As I said in the other issue, a two-page wizard is an anti-pattern. The app creation form is already very shallow, so it should be no problem to add it there at all.

johanstokking, 2019-09-02:
As mentioned offline, the goal is to replace this manual linking step by the magic. If there is an AS in the cluster, the only thing we need is a checkbox.

Whether or not we put that in a wizard is unrelated to this issue.

htdvisser, 2019-09-02:
> a two-page wizard is an anti-pattern

Right now it might be two steps, but I can imagine that we may have more steps in the future:

- Configuring application-level payload encoders/decoders
- Creating an API key for reading/writing traffic with MQTT
- Creating an API key for an integration and setting up that integration

> Whether or not we put that in a wizard is unrelated to this issue.

My comment was about the *How do you propose to implement this?* section. A checkbox is one way to implement it, a wizard is another way to implement it. Therefore I don't think the comment was unrelated to the issue.

kschiffer, 2019-09-02:
> Right now it might be two steps, but I can imagine that we may have more steps in the future:

I see that but it doesn't change my reasoning. We can move to a wizard as soon as the use-case matches.

> Therefore I don't think the comment was unrelated to the issue.

I agree.

PR #1316 (2019-09-05): Add option to automatically link apps on creation
#### Summary
Closes #1282 
[image]

#### Changes
- Add fields for linking the app automatically
- Add logic to submit logic to create an api key and use it to link the app
- Refactor some messages

#### Notes for Reviewers
It would be good to also check whether the current user has the right to link applications, but I would like to do that via #1200. For now, it only checks whether the AS is in the cluster.

#### Release Notes
Added option to link applications automatically upon creation
closed 2019-09-05 by kschiffer