TheThingsNetwork/lorawan-stack #1302: Resets join nonces not effective (+ activation layout fixes)

johanstokking, 2019-09-03:
#### Summary

The **Resets join nonces** setting is not effective in the Console.

#### Steps to Reproduce

1. Create new device, check **resets join nonces**
2. Go to the device after creation, the **resets join nonces** is unchecked
3. Check again, save
4. Go to device, still unchecked

Also, no call to JS seems to be made in step 3

#### What do you see now?

Ineffective setting of `resets_join_nonces`. Also, not sure if this is a layout/rendering issue, but it would be nice to slightly reorder the form:

[image]

#### What do you want to see instead?

Effective setting of `resets_join_nonces`.

As for the layout:
- AppKey before NwkKey
- "Join EUI" to become "JoinEUI"
- "Device EUI" to become "DevEUI"
- In text under input boxes:
   - "The connected Join EUI" to become "JoinEUI identifies the Join Server (in LoRaWAN 1.0.x known as AppEUI)"
   - "The device EUI ..." to become "The DevEUI ..."
   - "... Network Key" to become "... NwkKey"
   - "The App Key ..." to become "The AppKey ..."

#### Environment

Latest master

#### How do you propose to implement this?

Not sure

Text is probably editing some translations

#### Can you do this yourself and submit a Pull Request?

Can review

kschiffer, 2019-09-03:
Reassigning to @bafonins, as he worked on the device form recently.

johanstokking, 2019-09-03:
```
diff --git a/pkg/webui/console/components/device-data-form/messages.js b/pkg/webui/console/components/device-data-form/messages.js
index 3e836c946..43a2fb2bd 100644
--- a/pkg/webui/console/components/device-data-form/messages.js
+++ b/pkg/webui/console/components/device-data-form/messages.js
@@ -19,19 +19,20 @@ export default defineMessages({
   createDevice: 'Create Device',
   deviceIdPlaceholder: 'my-new-device',
   deviceNamePlaceholder: 'My New Device',
-  deviceIdDescription: 'Unique Identifier of your device; this cannot be changed afterwards',
+  deviceIdDescription: 'ID of your device; this cannot be changed afterwards',
   deviceNameDescription: 'Human friendly name of your device for display purposes',
   deviceDescDescription:
     'Optional device description; can also be used to save notes about the device',
-  joinEUIDescription: 'The connected Join EUI',
+  joinEUIDescription: 'The JoinEUI identifies the Join Server (or LoRaWAN 1.0.x AppEUI)',
   leaveBlankPlaceholder: 'Leave blank to generate automatically',
   resetsJoinNonces: 'Resets Join Nonces',
   resetsFCnt: 'Resets Frame Counters',
-  deviceEUIDescription: 'The device EUI is the unique identifier for this device on the network.',
-  nwkKeyDescription: 'The encrypted Network Key',
+  deviceEUIDescription: 'The DevEUI is the unique identifier for this device on the network.',
+  nwkKeyDescription:
+    'The root key to derive network session keys to secure communication between the device and the network',
   appKeyDescription:
-    'The App Key is used to secure the communication between your device and the network.',
-  appSKeyDescription: 'App Session Key',
+    'The root key to derive the application session key to secure communication between the device and the application',
+  appSKeyDescription: 'Application Session Key',
   fwdNtwkKeyDescription: 'Forwarding Network Session Integrity Key (or LoRaWAN 1.0.x NwkSKey)',
   sNtwkSIKeyDescription: 'Serving Network Session Integrity Key (only for LoRaWAN 1.1+)',
   ntwkSEncKeyDescription: 'Network Session Encryption Key (only for LoRaWAN 1.1+)',
```

bafonins, 2019-09-04:
Moved layout fixes to a separate issue, see https://github.com/TheThingsNetwork/lorawan-stack/issues/1304

closed 2019-09-04 by bafonins