TheThingsNetwork/lorawan-stack #1303: Support Concentrator Design v2 Fields in Basic Station config

KrishnaIyer, 2019-09-03:
#### Summary

Support Concentrator Design v2 Fields in Basic Station config. Related to #761.

#### Why do we need this?

Support gateways that follow the [v2 reference design](https://doc.sm.tc/station/gw_v2.html).

#### What is already there? What do you see now?

Support only for v1 and v1.5 ref design.

#### What is missing? What do you want to see?

Support v2 ref design.

#### Environment

Basic Station

#### How do you propose to implement this?

- First support multiple SX1301Conf (#761 )
- Add additional fields to the `router_config` message.
- Testing with a v2 gateway.

#### Can you do this yourself and submit a Pull Request?

Yes

cc: @johanstokking

johanstokking, 2019-09-03:
Adding `compat/api` as we're probably extending `Gateway` entity

johanstokking, 2019-09-10:
Blocked by #761

htdvisser, 2020-08-24:
Let's keep in backlog until there is actual demand for this.

rish-c, 2021-01-11:
Can we add bump this up after the conference? Let me know if we need access to a V2 ref gateway for testing.

htdvisser, 2021-01-11:
@rish please indicate which / how many customers are asking for this, and why they need it, so that we can understand the demand, and prioritize accordingly.

htdvisser, 2021-03-01:
This is blocked on receiving v2 gateways for testing. @rish-c and @ajovertoom are working on that as far as I know. Please update this issue when we have the gateways.

ajovertoom, 2021-03-04:
@KrishnaIyer our spare Cisco IXM is coming to the office once Sharan has visited storage this week.

KrishnaIyer, 2021-04-30:
Bumping the issue due to demand

KrishnaIyer, 2021-05-21:
The basic changes are already covered in #4190 . This should already be working with v2 ref gateways. I'll keep this issue open until we can confirm with testing.

KrishnaIyer, 2021-06-03:
No response for our request to test yet. Moving this to the next milestone.

KrishnaIyer, 2021-09-15:
We have no response for the corresponding gateway vendor. We have no chance to test so closing this.

closed 2021-09-15 by KrishnaIyer