TheThingsNetwork/lorawan-stack #1315: Documentation on how to join TTN gateway to TTN stack

loganmc10, 2019-09-04:
#### Summary

I currently have a TTN gateway (connected to the public TTN network). I would like to connect to to a private TTN v3 stack. I cannot find any documentation on how to do this.

#### Why do we need this?

TTN gateway is a great learning device, as is the TTN stack. It would be good to show how these devices can work together.

#### What is already there? What do you see now?

No current documentation on this.

#### What is missing? What do you want to see?

Documentation included with the TTN stack on how to join a TTN gateway to the network server

#### Environment

Ubuntu

#### How do you propose to implement this?

Addition to the Getting Started Guide

#### Can you do this yourself and submit a Pull Request?

No, not sure how to join a TTN gateway to a private network server

johanstokking, 2019-09-05:
@loganmc10 thanks for filing the issue and showing interest in combining The Things Indoor Gateway with The Things Stack V3.

It is definitely possible, but it requires some steps to claim the gateway so that the configuration endpoint the gateway connects to (CUPS) knows which network server (LNS) to communicate to the gateway. Today, we route all traffic from TTIGs to the public community network US and EU clusters.

This indeed needs documentation, as well as probably some user friendly way to claim gateways. We have an open issue for this, set for this month, see https://github.com/TheThingsNetwork/lorawan-stack/issues/1140. Therefore I'm marking this issue as duplicate. Please upvote the aforementioned issue to show support!

johanstokking, 2019-09-05:
Duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/1140

johanstokking, 2019-09-05:
@loganmc10 for the record, which are you referring to the original The Things Gateway (white diamond shape), or the new The Things Indoor Gateway (white smaller thing)?

loganmc10, 2019-09-05:
I'm referring to the original Things Gateway, thanks for getting back to me

closed 2019-09-05 by loganmc10