TheThingsNetwork/lorawan-stack #1319: Improve error rendering in Web UI

htdvisser, 2019-09-05:
#### Summary

We need to improve how errors are rendered in the web UI.

#### Why do we need this?

Right now they're not informative enough.

#### What is already there? What do you see now?

Creating an entity (in this case an organization) with an invalid ID returns the following error stack in the CLI:

```
error:pkg/errors:validation (invalid `organization`: embedded message failed validation)
    name=CreateOrganizationRequestValidationError
    reason=embedded message failed validation
    field=organization
    correlation_id=ff913a0396cc4b109c869cf4d172c62d
--- error:pkg/errors:validation (invalid `ids`: embedded message failed validation)
    field=ids
    name=OrganizationValidationError
    reason=embedded message failed validation
--- error:pkg/errors:validation (invalid `organization_id`: value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$")
    reason=value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$"
    field=organization_id
    name=OrganizationIdentifiersValidationError
```

but the console only shows

[image]

#### What is missing? What do you want to see?

I'd like to see what caused the error.

#### How do you propose to implement this?

Look at the `cause` field in the response:

```json
{
  "code": 3,
  "message": "error:pkg/errors:validation (invalid `organization`: embedded message failed validation)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/errors",
      "name": "validation",
      "message_format": "invalid `{field}`: {reason}",
      "attributes": {
        "field": "organization",
        "name": "CreateOrganizationRequestValidationError",
        "reason": "embedded message failed validation"
      },
      "correlation_id": "eefb90988c284758956cc3585218ea7f",
      "cause": {
        "namespace": "pkg/errors",
        "name": "validation",
        "message_format": "invalid `{field}`: {reason}",
        "attributes": {
          "field": "ids",
          "name": "OrganizationValidationError",
          "reason": "embedded message failed validation"
        },
        "cause": {
          "namespace": "pkg/errors",
          "name": "validation",
          "message_format": "invalid `{field}`: {reason}",
          "attributes": {
            "field": "organization_id",
            "name": "OrganizationIdentifiersValidationError",
            "reason": "value does not match regex pattern \"^[a-z0-9](?:[-]?[a-z0-9]){2,}$\""
          },
          "code": 3
        },
        "code": 3
      },
      "code": 3
    }
  ]
}
```

pgalic96, 2019-10-03:
@htdvisser I am trying to replicate this issue but it seems that these invalid ID's are always caught by the frontend field validation. How did you manage to get past the validation?

htdvisser, 2019-10-03:
Maybe the validation has been fixed since this issue was created, but that doesn't change the fact that nested errors (which can come from different places) aren't displayed properly.

kschiffer, 2019-10-08:
Generally speaking, the errors returned from the backend are not really user-friendly and often require developer-knowledge (e.g. hinting towards a value not matching a certain regex pattern). Whenever possible, we should, therefore, try to "translate" backend errors to something that assists the user appropriately in resolving the problem.

I think it makes sense to extend our error notification to have a collapsible "Error Details" section, which contains the raw error message returned from the backend. We could then display a user-friendly message (e.g. "There was a problem while creating the End Device", or provide just the cause message, see #1426) and provide developer-directed details on demand.

Also, we should generally ensure that avoidable validation errors are being caught on the frontend before submit.

htdvisser, 2019-10-08:
> the errors returned from the backend are not really user-friendly
> [...]
> we should, therefore, try to "translate" backend errors to something that assists the user

This may be a bit out of scope for this issue but yes. Typically this is the goal of the first (outermost) error in the chain. We need to improve those.

Then we could make the console recognize certain errors (like this `error:pkg/errors:validation`) error.

> we should generally ensure that avoidable validation errors are being caught on the frontend before submit.

Sure, but also out of scope for this issue.

kschiffer, 2019-10-08:
So, would you agree with adding "Error Details" to our error notifications, as an intermediate solution?

PR #1515 (2019-10-31): Display backend error details in console

#### Summary

Closes #1319 

#### Changes

- Change message rendering in `Notification` component
- Create new `ErrorNotification` component that wraps `Notification`
- Add new `Details` component that displays error details if available
- Update existing `Notification` components
closed 2019-10-31 by pgalic96