TheThingsNetwork/lorawan-stack #1327: Change container width constraint

bafonins, 2019-09-06:
#### Summary

[image]
[image]
[image]

#### Why do we need this?

Dont jam table columns when possible.
We should utilize all available space.

#### What is already there? What do you see now?

* Table components
* Container constraints on list pages

#### What is missing? What do you want to see?

Something like this:
[image]

#### Environment

browser (console)

#### How do you propose to implement this?

Change responsive grid sizes/constraints

#### Can you do this yourself and submit a Pull Request?

can review

kschiffer, 2019-09-16:
> We should utilize all available space.

This was fixed via #1330.

> Dont jam table columns when possible.

Will do. Needs some planning via #1172.

kschiffer, 2020-07-27:
Closing this since it has since been resolved. Table columns won't jam anymore since tables have a min-width in smaller screensizes and enable scrolling. More improvements will come for event views via #2893

closed 2020-07-27 by kschiffer