TheThingsNetwork/lorawan-stack #1328: Support LoRa Cloud Device Management

johanstokking, 2019-09-06:
#### Summary

Support LoRa Cloud Device Management (and plumbing for application packages now that we need it)

#### Why do we need this?

To natively support the Semtech modems

#### What is already there? What do you see now?

No support, only some ideas about supporting application-layer services (called application packages) based on FPort. These ideas have not been written down, so I'll do that in this issue.

#### What is missing? What do you want to see?

##### Supporting application packages

The Application Server should get a pluggable and dynamic table of "application packages" per `FPort`. These packages provide functionality on the end device and on the server.

>Example application packages are time synchronization, remote multicast setup, data fragmentation, etc.

Application packages are:
1. MAC layer independent. However, they need to be aware of the maximum (downlink) payload size
   - @rvolosatovs please comment on how to communicate this between NS-AS. My requirement here is that AS is not going to compute that based on DR and PHY; NS should do that. My proposal is to carry this in `ApplicationUp` to keep things streaming
2. Region independent
3. Versioned (as in, each version is essentially a separate implementation)
4. For application traffic, independent of other packages
5. Stateful. They need one (abstract) way to store state. I would suggest a common registry for application packages to store binary data (marshaled from end device state protos per package)
6. Be able to publish messages, triggering the AS to publish a new type of `ApplicationUp.Up` message with "the payload" (whatever that may be) of an application package. We need to keep this very generic, so a proto `Struct` would suffice
7. Be able to schedule downlink messages
8. In the future, be able to expose gRPC services

See Application Layer Package Design Guidelines (TR003) [1]

Packages are registered on an `FPort`; there can only be one package implementation per port. This registration is dynamic and can fluctuate per device during its lifetime by user configuration (through device registry) or through other packages.

There is a special package ("Multi-package access protocol" [2]) that is itself fixed on `FPort` 225, but can discover packages and register them for the concerning end device, i.e. configure "remote multicast setup V2" on `FPort` 223.

So, each `EndDevice` gets a mapping of `FPort` to some versioned package identifier, that is stored in Application Server.

In the future, there's a default mapping per end device coming from the Device Repository. References #263 

- [ ] Plumbing for application packages with registration by (versioned) identifier, interface, storage in API, CLI support for management

Resources:
[1] Application Layer Package Design Guidelines (TR003). LoRa Alliance confidential, see linked issue
[2] Multi-package access protocol. LoRa Alliance confidential, see linked issue

##### LoRa Cloud Device Management support

This is essentially one application package that's registered on `FPort` 200.

See https://www.loracloud.com/documentation/device_management. We don't need to implement this in its entirety yet, but at least:

- [ ] Token management ([see here](https://www.loracloud.com/documentation/device_management?url=v1.html#token-management)). The way this seems to work is that the device owner gives a token to the Application Server to authenticate on behalf of the device owner. There is one token for a group of devices. I think this fits naturally with the concept of an application with many devices, i.e. store the LoRa Cloud Device Management token on the application level
- [ ] ~Device Management~. This is not necessary (yet) as devices will be automatically provisioned when sending uplink messages
- [ ] Sending uplink messages ([see here](https://www.loracloud.com/documentation/device_management?url=v1.html#deliver-uplink-message)). So once we have that application packages registration per device in place, we'll automatically load this Semtech modem package on `FPort` 200 and the package sends the uplink
- [ ] Act on uplink responses ([see here](https://www.loracloud.com/documentation/device_management?url=v1.html#uplink-response)). We get this response when sending uplink messages
   - [ ] The package should publish messages (see 6 above), whenever something interesting has been received (like a file, stream records, fulfilled requests, info fields and log messages). We can probably just serialize relevant fields from the Uplink Response in a key/value struct
   - [ ] Schedule downlink messages that were sent as part of the Uplink Response
- [ ] Documentation for modem users

#### How do you propose to implement this?

As for the application packages, I would suggest a very simple interface for packages and that packages can register themselves. I'm well aware of the naming collision of Go packages and LoRaWAN application packages, but I'm afraid we have to deal with that in our codebase.

As for the LoRa Cloud Device Management service, that is the first package implementation that needs an identifier, like `lora-cloud-device-management-v1`. It's registered on `FPort` 200, processes uplink messages, sends them to LoRa Cloud, and acts on responses (publishing package payload that AS publishes as a new `ApplicationUp`, as well as scheduling downlink messages).

#### Can you do this yourself and submit a Pull Request?

Can review and provide input for discussion. Please mention me when there are any questions or if you want feedback.

Please remove `discussion` label when:
1. Everything is clear
2. @rvolosatovs provided an idea on how to communicate constraints for the application-layer (like max downlink payload size) from NS to AS

johanstokking, 2019-09-06:
I think we should refactor this part; https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/applicationserver/linking.go#L229-L252

- Add a `sendUp(*io.ContextualApplicationUp)` to `link`
- Call `sendUp()` in `handleUp()` instead, also do the cleaning of join-accept and filtering of messages that we don't send upstream (like downlink queue invalidation) there
- Call `sendUp()` from goroutines after running uplink messages through application package implementations (in `handleUplink()`)

adriansmares, 2019-09-13:
Regarding the mapping between and `EndDevice` and active application packages - do we want to extend `EndDevice` with this mapping or we will have a separate mapping registry ? Same for storage: do we store the package-specific data inside the device or in a separate registry ? The trade-off here is between having to create the entry in the registries manually (adding a new step to device creation) vs. making `EndDevice`s way too big. cc @johanstokking

rvolosatovs, 2019-09-13:
As a first step, I think we should include `capacity` in details of `errApplicationDownlinkTooLong` sent https://github.com/TheThingsNetwork/lorawan-stack/blob/e69e3e13562e0ac48bf6adeaf4eb69640ef9195c/pkg/networkserver/downlink.go#L155-L165 when application downlink is too long and cannot be scheduled. This is already enough for AS to be able to "fixup" the downlink, i.e.:
1. AS enqueues downlink with arbitrary, but realistic(i.e. less that 255 bytes in all cases, probably account for 16 bytes of FOpts as well) payload length
2. If NS fails the downlink - split downlink into several downlinks, use the capacity returned to determine the split strategy
3. Go to 1.

I believe something like this should work for most cases.

As the next step I think we should add a new application uplink:
```protobuf
// PayloadLimits represent the maximum length of application payload that Network Server can schedule at the next downlink opportunity.
message PayloadLimits {
	uint32 Rx1 = 1;
	uint32 Rx2 = 2;
}
```
This uplink should be sent to AS every time the respective limits change(i.e. an uplink or a registry update, which may influence this)

Other than that I think NS shall expose
```protobuf
service AsNs {
	// Other RPCs...
	rpc GetPayloadLimits(EndDeviceIndentifiers) returns (PayloadLimits)
```
Which is necessary for e.g. multicast devices. AS may also use this if this information is lost.

Note, that `capacity` returned as details of possible downlink failure is different from the `PayloadLimits`.
The Rx1/Rx2 values in the `PayloadLimits` represent the maximum possible `FRMPayload` length(i.e. including `FOpts` length). It's then AS's responsibility to use the data and decide to "risk it" or not, by not accounting for up to 16 bytes of `FOpts`.
`capacity` in error details, however, is the maximum payload length *for the downlink attempted*

johanstokking, 2019-09-19:
> Edit: My proposal would look like this: https://github.com/TheThingsNetwork/lorawan-stack/blob/feature/1328-package-registry/api/applicationserver_packages.proto

This is great

johanstokking, 2019-09-19:
> As a first step, I think we should include `capacity` in details of `errApplicationDownlinkTooLong` sent
> 
> [...]
>
> when application downlink is too long and cannot be scheduled. This is already enough for AS to be able to "fixup" the downlink, i.e.:

Afaik this is too much pingpong between NS and AS and becomes more of a trial-and-error approach. Also, we can throw it away when we implement your next proposal, which doesn't make it a good time investment.

> As the next step I think we should add a new application uplink:
>
> ```protobuf
> // PayloadLimits represent the maximum length of application payload that Network Server can schedule at the next [...]

I think this is the right and only approach to choose.

> Other than that I think NS shall expose
> 
> ```protobuf
> service AsNs {
> 	// Other RPCs...
> 	rpc GetPayloadLimits(EndDeviceIndentifiers) returns (PayloadLimits)
> }
> ```
> 
> Which is necessary for e.g. multicast devices. AS may also use this if this information is lost.

Yes

> The Rx1/Rx2 values in the `PayloadLimits` represent the maximum possible `FRMPayload` length(i.e. including `FOpts` length). It's then AS's responsibility to use the data and decide to "risk it" or [...]

AS is only interested in maximum allowed `FRMPayload` and in nothing else, indeed.

adriansmares, 2020-01-08:
Since #1399 added application packages support and #1751 adds support for the LoRa Cloud Device Management Service, I've extracted the remaining part of the issue into #1829 and I'm closing this one.

closed 2020-01-08 by adriansmares