TheThingsNetwork/lorawan-stack #1332: GS flaky test

adriansmares, 2019-09-07:
#### Summary

The GS test `TestGatewayServer/Traffic/grpc/Upstream/OneValidFS` is flaky.

#### Steps to Reproduce

1. Bad luck.

#### What do you see now?

```
  WARN No cookie hash key configured, generated a random one hash_key=457866FDAC82E4F6FE767B3BE0239E6FC23C86B3CFDFDE0EBFEDE3424DC57C9EA0AEB292DD70C9627AFA0856CA93D88D23EFF5C94E6CC0E9FC9C30BE92C4FDB1 namespace=web test_name=TestGatewayServer

  WARN No cookie block key configured, generated a random one block_key=E023A3DD664C88D722453E5BF39583AA6848D6225F7CC6BF6D65ADFAE204B9B7 namespace=web test_name=TestGatewayServer

  WARN No static assets found in any search path namespace=web search_paths=[] test_name=TestGatewayServer

  WARN No rights TTL configured                 test_name=TestGatewayServer

  WARN No cluster key configured, generated a random one key=9f200ad27f76ea38f39846be074ae9d26f3c0c99a397c3fb103c84b5936bd649 test_name=TestGatewayServer

  INFO Listening for connections                address=:9187 namespace=grpc protocol=gRPC test_name=TestGatewayServer

  INFO Finished unary call                      duration=1.816449ms grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ1VTG8W27HWZ214Y10FX test_name=TestGatewayServer

  INFO Connected                                gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:42248 request_id=01DM6GZ1VXFNKH41XPE83TQN9V test_name=TestGatewayServer

  INFO Finished streaming call                  duration=252.946636ms error=error:unknown:unknown (context canceled) error_correlation_id=618f3d6cdb044c64b967aa2c564e761e grpc_code=Canceled grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ1VXFNKH41XPE83TQN9V test_name=TestGatewayServer

  INFO Disconnected                             gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc remote_addr=127.0.0.1:42248 request_id=01DM6GZ1VXFNKH41XPE83TQN9V test_name=TestGatewayServer

  WARN Finished unary call                      duration=716.972µs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `eui-aaee000000000000`) error_correlation_id=d3b9fba8eebb49c9b24e160641a0c768 error_name=no_gateway_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ240EKW42S1B0YT6VWW3 test_name=TestGatewayServer

  WARN Finished unary call                      duration=552.973µs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `invalid-gateway`) error_correlation_id=15d5c92e2fde4c13b4f4968bad36aba7 error_name=no_gateway_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ244ZK92C2P1YD10CY1S test_name=TestGatewayServer

  INFO Finished unary call                      duration=128.259µs error=error:pkg/ttnpb:identifiers (invalid identifiers) error_cause=error:pkg/errors:validation (invalid `gateway_id`: value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$") error_correlation_id=4fb7557321d642da9c55e015af2a5245 error_name=identifiers error_namespace=pkg/ttnpb grpc_code=InvalidArgument grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ2468CXGMBTWDFTQ3XE7 test_name=TestGatewayServer

  INFO Finished unary call                      duration=121.967µs error=error:pkg/ttnpb:identifiers (invalid identifiers) error_cause=error:pkg/errors:validation (invalid `gateway_id`: value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$") error_correlation_id=e543ae0c02f94748a8b68ab31b01c750 error_name=identifiers error_namespace=pkg/ttnpb grpc_code=InvalidArgument grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ2479Q3B8NQFHVVTC5WT test_name=TestGatewayServer

  INFO Finished unary call                      duration=958.764µs error=error:pkg/gatewayserver:not_connected (gateway `eui-aaee000000000000` not connected) error_correlation_id=8e6cbbe1cb2742089f69f7dfb4d3b197 error_name=not_connected error_namespace=pkg/gatewayserver grpc_code=NotFound grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DM6GZ2CADKNW4EGJG5NM9BRK test_name=TestGatewayServer

  INFO Finished unary call                      duration=1.316157ms grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ2CDA3H6BH3S0E6Y1XF9 test_name=TestGatewayServer

  INFO Connected                                gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:42260 request_id=01DM6GZ2CF2D60SM4F81H4CYFF test_name=TestGatewayServer

  INFO Finished unary call                      duration=1.059837ms grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DM6GZ2MJVXM6F6HRMH8P03SF test_name=TestGatewayServer

  INFO Finished unary call                      duration=445.021µs grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DM6GZ2MMV4XXJDXXSTG8EWBZ test_name=TestGatewayServer

  INFO Finished unary call                      duration=584.997µs grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DM6GZ2MP027YZH18F8FF2V3T test_name=TestGatewayServer

  INFO Finished unary call                      duration=732.869µs grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DM6GZ2MRC3MJEAQGSGVVW3GN test_name=TestGatewayServer

  INFO Disconnected                             gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc remote_addr=127.0.0.1:42260 request_id=01DM6GZ2CF2D60SM4F81H4CYFF test_name=TestGatewayServer

  INFO Finished streaming call                  duration=266.656308ms error=error:unknown:unknown (context canceled) error_correlation_id=64038f82618d4202b647f714b2ec05c4 grpc_code=Canceled grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DM6GZ2CF2D60SM4F81H4CYFF test_name=TestGatewayServer

--- FAIL: TestGatewayServer (1.87s)

    --- FAIL: TestGatewayServer/Traffic/grpc (0.53s)

        --- FAIL: TestGatewayServer/Traffic/grpc/Upstream (0.27s)

            --- FAIL: TestGatewayServer/Traffic/grpc/Upstream/OneValidFSK (0.00s)

                gatewayserver_test.go:810: gs.up.forward event published

                doc.go:66: 

                    /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver_test.go:876

                    Expected: '2'

                    Actual:   '1'

                    (Should be equal)

FAIL

FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver	1.914s
```

#### What do you want to see instead?

The test should not fail randomly.

#### Can you do this yourself and submit a Pull Request?

Can review.

htdvisser, 2019-10-08:
Looks related:

```
  WARN No cookie hash key configured, generated a random one hash_key=2FECCF81EFFEF0C8110CCBE2BE8B23C26E337279E62756ACFE16375BB6897E3EBA5759D84E150BB7E17A99EBEDE387B096EE41BB4DD44BC19C9609EFC8328F06 namespace=web test_name=TestGatewayServer
  WARN No cookie block key configured, generated a random one block_key=822E142EF552CFEEC7E546D22F7E36ED26CBCE6D79A8C4415898EC242919D649 namespace=web test_name=TestGatewayServer
  WARN No static assets found in any search path namespace=web search_paths=[] test_name=TestGatewayServer
  WARN No rights TTL configured                 test_name=TestGatewayServer
  WARN No cluster key configured, generated a random one key=eaadd899806bb4c13450d4257d7c6110c592da1b724a8978068a8f9187eb53d9 test_name=TestGatewayServer
  INFO Listening for connections                address=:9187 namespace=grpc protocol=gRPC test_name=TestGatewayServer
  INFO Finished unary call                      duration=5.012417ms grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DPN4H70FPFWJRBCCZAJTAS4W test_name=TestGatewayServer
  INFO Connected                                gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:41030 request_id=01DPN4H70SHSF2C64N1HDPVRJJ test_name=TestGatewayServer
  INFO Finished streaming call                  duration=245.644341ms error=error:unknown:unknown (context canceled) error_correlation_id=1257f760d52f4150a71f3ee60703974a grpc_code=Canceled grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DPN4H70SHSF2C64N1HDPVRJJ test_name=TestGatewayServer
  WARN Finished unary call                      duration=604.26Âµs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `eui-aaee000000000000`) error_correlation_id=46b6a81d7eeb4c089f7e3c4eedc841f5 error_name=no_gateway_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DPN4H78GTQJD37YRDRTY6Y9V test_name=TestGatewayServer
  WARN Finished unary call                      duration=503.568Âµs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `invalid-gateway`) error_correlation_id=eb72c1e0978a4702b6bc36439c272e00 error_name=no_gateway_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DPN4H78JP2MV2CWX8NVV1848 test_name=TestGatewayServer
  INFO Disconnected                             gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc remote_addr=127.0.0.1:41030 request_id=01DPN4H70SHSF2C64N1HDPVRJJ test_name=TestGatewayServer
  INFO Finished unary call                      duration=113.698Âµs error=error:pkg/ttnpb:identifiers (invalid identifiers) error_cause=error:pkg/errors:validation (invalid `gateway_id`: value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$") error_correlation_id=efc5641d14db4e939380d3daefa580bb error_name=identifiers error_namespace=pkg/ttnpb grpc_code=InvalidArgument grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DPN4H78M598DAMETTSJ2FZBP test_name=TestGatewayServer
  INFO Finished unary call                      duration=98.372Âµs error=error:pkg/ttnpb:identifiers (invalid identifiers) error_cause=error:pkg/errors:validation (invalid `gateway_id`: value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$") error_correlation_id=55e3f96c3e1e4744a74efa5eb9fd426a error_name=identifiers error_namespace=pkg/ttnpb grpc_code=InvalidArgument grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DPN4H78NJW6NKZKT4H6YX47J test_name=TestGatewayServer
  INFO Finished unary call                      duration=765.197Âµs error=error:pkg/gatewayserver:not_connected (gateway `eui-aaee000000000000` not connected) error_correlation_id=b07dd87e51824fc29bba3c808e38a183 error_name=not_connected error_namespace=pkg/gatewayserver grpc_code=NotFound grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DPN4H7GQBS44PQD1EWMJ90MV test_name=TestGatewayServer
  INFO Finished unary call                      duration=863.468Âµs grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01DPN4H7GS4ZEYAXTN31QZ4WJ9 test_name=TestGatewayServer
  INFO Connected                                gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:41042 request_id=01DPN4H7GT1NBTJKJCGEMN4QKT test_name=TestGatewayServer
  INFO Finished unary call                      duration=580.106Âµs grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DPN4H7RW3SJ569D1GQZ7HF3B test_name=TestGatewayServer
  INFO Finished unary call                      duration=655.422Âµs grpc_method=GetGatewayConnectionStats grpc_service=ttn.lorawan.v3.Gs namespace=grpc request_id=01DPN4H7RY0P6F19NTX7W9P5ZE test_name=TestGatewayServer
[... cut ...]
        --- FAIL: TestGatewayServer/Traffic/grpc/Upstream (0.26s)
            --- FAIL: TestGatewayServer/Traffic/grpc/Upstream/OneValidLoRa (0.00s)
                gatewayserver_test.go:812: gs.up.forward event published
                doc.go:66: 
                    /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver_test.go:878
                    Expected: '1'
                    Actual:   '0'
                    (Should be equal)
FAIL
FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver	1.915s
```

rvolosatovs, 2020-02-20:
Look like I found another one:
```
  WARN No cookie hash key configured, generated a random one hash_key=AF8CAC67C022D6738C91ACEC222EB0FEEB1EED3075DCC91EB38660FB9C8C05B72CD2CCBA4A6CE81DDDD937D6C1C86AE10ACE17BB57D6F44BEC7114FC36055BD7 namespace=web test_name=TestGatewayServer

  WARN No cookie block key configured, generated a random one block_key=D0DA3B8FF79A7F4A35AF7199A95465A5DA4E99D5EB7B8FEEF7A2A4CD3B4D2BC8 namespace=web test_name=TestGatewayServer

  WARN No static assets found in any search path namespace=web search_paths=[] test_name=TestGatewayServer

  WARN No rights TTL configured                 test_name=TestGatewayServer

  WARN No cluster key configured, generated a random one key=f6d812ec6397f4cd4b12e7725ca88d99334ada3fe94e5b1181b5b8101f48d66d test_name=TestGatewayServer

  INFO Listening for connections                address=:9187 namespace=grpc protocol=gRPC test_name=TestGatewayServer

  INFO Finished unary call                      duration=2.932065ms grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1H2JKQ3YDTN36EFNCV0CS56 test_name=TestGatewayServer

  INFO Connected                                gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:36532 request_id=01E1H2JKQ7T0A088T8FZY72T0W test_name=TestGatewayServer

  WARN Finished streaming call                  duration=247.357924ms error=context deadline exceeded grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1H2JKQ7T0A088T8FZY72T0W test_name=TestGatewayServer

  WARN Link failed                              error=rpc error: code = DeadlineExceeded desc = context deadline exceeded gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc remote_addr=127.0.0.1:36532 request_id=01E1H2JKQ7T0A088T8FZY72T0W test_name=TestGatewayServer

  INFO Disconnected                             gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:36532 request_id=01E1H2JKQ7T0A088T8FZY72T0W test_name=TestGatewayServer

  WARN Task failed                              error=mux: listener closed invocation=1 namespace=gatewayserver task_id=serve_mqtt/:1882 test_name=TestGatewayServer

--- FAIL: TestGatewayServer (4.66s)

    --- FAIL: TestGatewayServer/Authenticate/grpc (0.26s)

        --- FAIL: TestGatewayServer/Authenticate/grpc/ValidIDAndKey (0.26s)

            gatewayserver_test.go:559: Expected deadline exceeded with valid auth, but have error:unknown:unknown (context deadline exceeded)

  WARN Task failed                              error=read udp [::]:1700: use of closed network connection invocation=1 namespace=gatewayserver task_id=serve_udp/:1700 test_name=TestGatewayServer

  WARN Task failed                              error=http: Server closed invocation=1 namespace=gatewayserver task_id=serve_basicstation/:1887 test_name=TestGatewayServer

FAIL

  WARN transport: http2Server.HandleStreams failed to read frame: io: read/write on closed pipe namespace=grpc test_name=TestGatewayServer

coverage: 41.4% of statements

FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver	4.714s
```

rvolosatovs, 2020-02-21:
And another one with `basicstation` this time:

<details>
<summary>Details</summary>

  WARN No cookie hash key configured, generated a random one hash_key=686D148310C51B627BF34558528A47B84D1001620F48956E6B042A7CCBC82CC5B3F7AA873BA65BD7C7B43AEC7DA75CEE8F82761886A14AAF7182B394E1513879 namespace=web test_name=TestGatewayServer

  WARN No cookie block key configured, generated a random one block_key=A52935E61E7592777715B8CB46FAF8E3221A0DAC214BDD6A23607E668F3C7BAA namespace=web test_name=TestGatewayServer

  WARN No static assets found in any search path namespace=web search_paths=[] test_name=TestGatewayServer

  WARN No rights TTL configured                 test_name=TestGatewayServer

  WARN No cluster key configured, generated a random one key=965535d0e2ca0d5883c0e8c0fa0eef8fe4d9f97f25b7458e77dfa6b3cd15d5e0 test_name=TestGatewayServer

  INFO Listening for connections                address=:9187 namespace=grpc protocol=gRPC test_name=TestGatewayServer

  INFO Finished unary call                      duration=14.054791ms grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1KP9X43KPMHN775400EDVSY test_name=TestGatewayServer

  INFO Connected                                gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:60682 request_id=01E1KP9X4J3M4X91F397PTWGHG test_name=TestGatewayServer

  INFO Finished streaming call                  duration=239.767717ms error=context canceled grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1KP9X4J3M4X91F397PTWGHG test_name=TestGatewayServer

  INFO Disconnected                             gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:60682 request_id=01E1KP9X4J3M4X91F397PTWGHG test_name=TestGatewayServer

  WARN Finished unary call                      duration=566.11µs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `eui-aaee000000000000`) error_correlation_id=35b22abf384e4bf69c03e09931b4e689 error_name=no_gateway_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1KP9XC31MX21ZTFNCYARFMD test_name=TestGatewayServer uid=eui-aaee000000000000

  WARN Finished unary call                      duration=444.016µs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `invalid-gateway`) error_correlation_id=248c07bcb42b45cd951f72d683cf3804 error_name=no_gateway_rights error_namespace=pkg/auth/rights grpc_code=PermissionDenied grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1KP9XC5152VJWPBT2F7TB8E test_name=TestGatewayServer uid=invalid-gateway

  INFO Finished unary call                      duration=173.105µs error=error:pkg/ttnpb:identifiers (invalid identifiers) error_cause=error:pkg/errors:validation (invalid `gateway_id`: value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$") error_correlation_id=eed7ec8dbca0442d8ab04037d4f1b217 error_name=identifiers error_namespace=pkg/ttnpb field=gateway_id grpc_code=InvalidArgument grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs name=GatewayIdentifiersValidationError namespace=grpc reason=value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$" request_id=01E1KP9XC71EYJ5KJSGZWQZYJZ test_name=TestGatewayServer

  INFO Finished unary call                      duration=81.663µs error=error:pkg/ttnpb:identifiers (invalid identifiers) error_cause=error:pkg/errors:validation (invalid `gateway_id`: value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$") error_correlation_id=85ad95d2b6d046fc9fd1ab3c5aa3d17b error_name=identifiers error_namespace=pkg/ttnpb field=gateway_id grpc_code=InvalidArgument grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs name=GatewayIdentifiersValidationError namespace=grpc reason=value does not match regex pattern "^[a-z0-9](?:[-]?[a-z0-9]){2,}$" request_id=01E1KP9XC8JVRETETN2B9HYEAE test_name=TestGatewayServer

  INFO Finished unary call                      duration=923.455µs grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1KP9XMAVY68WF94PTMENQXW test_name=TestGatewayServer

  INFO Connected                                gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:60692 request_id=01E1KP9XMCPBPQ2Z859GA9PGJ8 test_name=TestGatewayServer

  INFO Finished unary call                      duration=1.109666ms grpc_method=GetConcentratorConfig grpc_service=ttn.lorawan.v3.GtwGs namespace=grpc request_id=01E1KP9XWAD7DW8MBEZGR97BXR test_name=TestGatewayServer

  WARN Disconnect existing connection           gateway_uid=eui-aaee000000000000 grpc_method=LinkGateway grpc_service=ttn.lorawan.v3.GtwGs namespace=gatewayserver/io/grpc protocol=grpc remote_addr=127.0.0.1:60694 request_id=01E1KP9XWCSB4Q9YMKQSE7KT4D test_name=TestGatewayServer

  WARN Finished streaming call                  duration=257.213855ms error=error:pkg/gatewayserver:new_connection (new connection from same gateway) error_correlation_id=b652b79bda0f404c875e3da58601241c error_name=new_connection error_namespace=pkg/gat [... cut ...]
                doc.go:66: 

                    /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver_test.go:923

                    Expected: '2'

                    Actual:   '1'

                    (Should be equal)

  WARN transport: http2Server.HandleStreams failed to read frame: io: read/write on closed pipe namespace=grpc test_name=TestGatewayServer

FAIL

coverage: 78.1% of statements

FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver	11.830s

</details>

rvolosatovs, 2020-02-26:
And yet another one, refs #2003 

<details>
<summary>Log</summary>

```log
2020-02-26T12:09:37.2950440Z   WARN No cookie hash key configured, generated a random one hash_key=2F2F7C66987C8D056542955EF87C2E847403300EB8902617A2F10CAB64A0BF43E65A02381BB8E70DD428A5BBBF6A5CA9D97F57D086553B7AC354335BF2CC2557 namespace=web test_name=TestClientTokenAuth
2020-02-26T12:09:37.2951747Z   WARN No cookie block key configured, generated a random one block_key=7A6A1716EA81C20770D8AE6606279EE50AA66BCEDA4BF45960D21F7457DD1908 namespace=web test_name=TestClientTokenAuth
2020-02-26T12:09:37.2952326Z   WARN No static assets found in any search path namespace=web search_paths=[] test_name=TestClientTokenAuth
2020-02-26T12:09:37.2952700Z   WARN No rights TTL configured                 test_name=TestClientTokenAuth
2020-02-26T12:09:37.2953277Z   WARN No cluster key configured, generated a random one key=4828de4e338f805fd9872013c35c32ab574805eeaabeead7b302be4142ad58fa test_name=TestClientTokenAuth
2020-02-26T12:09:37.2953659Z   INFO Listening for connections                address=:0 namespace=grpc protocol=gRPC test_name=TestClientTokenAuth
2020-02-26T12:09:37.2954826Z   INFO Finished unary call                      duration=25.801Âµs grpc_method=ListRights grpc_service=ttn.lorawan.v3.GatewayAccess namespace=grpc request_id=01E20PDKBNVNTJ15PBK9BHE31Y test_name=TestClientTokenAuth
2020-02-26T12:09:37.2955957Z   INFO Request handled                          duration=1.543711ms method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:58330 request_id=01E20PDKBMZ3TNAGG60ZA3C1FR response_size=0 status=200 test_name=TestClientTokenAuth url=/traffic/eui-0101010101010101
2020-02-26T12:09:37.2957459Z   INFO Finished unary call                      duration=11.5Âµs grpc_method=ListRights grpc_service=ttn.lorawan.v3.GatewayAccess namespace=grpc request_id=01E20PDKBPXGSGGN7JH07XH7J6 test_name=TestClientTokenAuth
2020-02-26T12:09:37.2958673Z   INFO Request handled                          duration=729.705Âµs method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:58332 request_id=01E20PDKBPM7HAKKRQHP2P2NEG response_size=0 status=200 test_name=TestClientTokenAuth url=/traffic/eui-0101010101010101
2020-02-26T12:09:37.2959790Z   INFO Finished unary call                      duration=12.9Âµs grpc_method=ListRights grpc_service=ttn.lorawan.v3.GatewayAccess namespace=grpc request_id=01E20PDKBQVGBE3GQGEAHEYBSE test_name=TestClientTokenAuth
2020-02-26T12:09:37.2960871Z   WARN Failed to connect                        endpoint=traffic error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `eui-0101010101010101`) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:58334 request_id=01E20PDKBQXW3VQ1QATA9B6BMS test_name=TestClientTokenAuth uid=eui-0101010101010101 url=/traffic/eui-0101010101010101
2020-02-26T12:09:37.2963193Z   INFO Request handled                          duration=630.705Âµs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `eui-0101010101010101`) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:58334 request_id=01E20PDKBQXW3VQ1QATA9B6BMS status=403 test_name=TestClientTokenAuth uid=eui-0101010101010101 url=/traffic/eui-0101010101010101
2020-02-26T12:09:37.2964592Z   INFO Request handled                          duration=273.002Âµs method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:58336 request_id=01E20PDKBRCEFWGBN88QV1VFDT response_size=0 status=200 test_name=TestClientTokenAuth url=/traffic/eui-0101010101010101
2020-02-26T12:09:37.2965683Z   INFO Finished unary call                      duration=12.1Âµs grpc_method=ListRights grpc_service=ttn.lorawan.v3.GatewayAccess namespace=grpc request_id=01E20PDKBST5716W4V4RBKHQG8 test_name=TestClientTokenAuth
2020-02-26T12:09:37.2967665Z   WARN Failed to connect                        endpoint=traffic error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `eui-1122334455667788`) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:58338 request_id=01E20PDKBRBFF9XF21D1WERF9R test_name=TestClientTokenAuth uid=eui-1122334455667788 url=/traffic/eui-1122334455667788
2020-02-26T12:09:37.2971339Z   INFO Request handled                          duration=525.004Âµs error=error:pkg/auth/rights:no_gateway_rights (no rights for gateway `eui-1122334455667788`) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:58338 request_id=01E20PDKBRBFF9XF21D1WERF9R status=403 test_name=TestClientTokenAuth uid=eui-1122334455667788 url=/traffic/eui-1122334455667788
2020-02-26T12:09:37.2972518Z   WARN transport: http2Server.HandleStreams failed to read frame: io: read/write on closed pipe namespace=grpc test_name=TestClientTokenAuth
2020-02-26T12:09:37.2973051Z   WARN No cookie hash key configured, generated a random one hash_key=D834925B52BC94223C4E7CA1FE9B6C83BE9EA2EBB03F5724EEBEAE1B05454FE5D5299CBE7984154F63003120EB536D2789904F1598614566E2F016617E672B0C namespace=web test_name=TestDiscover
2020-02-26T12:09:37.2973480Z   WARN No cookie block key configured, generated a random one block_key=D93A7217E0F0D4278F38BB522C167599F5130449C112D8FFECB3F3B0D18CC318 namespace=web test_name=TestDiscover
2020-02-26T12:09:37.2973841Z   WARN No static assets found in any search path namespace=web search_paths=[] test_name=TestDiscover
2020-02-26T12:09:37.2974085Z   WARN No rights TTL configured                 test_name=TestDiscover
[... cut ...]
2020-02-26T12:09:37.3018139Z 2020/02/26 12:09:23 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020-02-26T12:09:37.3018709Z --- FAIL: TestPingPong (0.35s)
2020-02-26T12:09:37.3062032Z ##[error]    basicstationlns_test.go:1352: Received server ping
2020-02-26T12:09:37.3066025Z ##[error]    basicstationlns_test.go:1367: Server pong timeout
2020-02-26T12:09:37.3066374Z FAIL
2020-02-26T12:09:37.3067122Z FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver/io/basicstationlns	0.937s
```

</details>

neoaggelos, 2020-03-03:
One more: https://github.com/TheThingsNetwork/lorawan-stack/pull/2036/checks?check_run_id=475381911

```log
ERROR Request error                            duration=65.476945ms error=error:pkg/web/middleware:http_recovered (Internal Server Error) method=GET namespace=gatewayserver/io/basicstation panic=concurrent write to websocket connection remote_addr=127.0.0.1:41304 request_id=01E26E87M66F6VC188T4N2VFRF status=500 test_name=TestPingPong url=/traffic/eui-0101010101010101
2020/02/28 17:42:05 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
  WARN transport: http2Server.HandleStreams failed to read frame: io: read/write on closed pipe--- FAIL: TestPingPong (0.34s)
##[error]    basicstationlns_test.go:1352: Received server ping
##[error]    basicstationlns_test.go:1367: Server pong timeout
FAIL
FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver/io/basicstationlns	0.934s
```

johanstokking, 2020-03-12:
This one might be serious as it may panic in production as well:

```
goroutine 1682 [running]:
go.thethings.network/lorawan-stack/pkg/ttnpb.encodeVarintGateway(0xabffad0, 0xcb, 0xcb, 0x0, 0xc, 0x0, 0xcb)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:4225 +0x1b6
go.thethings.network/lorawan-stack/pkg/ttnpb.(*GatewayConnectionStats).MarshalToSizedBuffer(0xa94f380, 0xabffad0, 0xcb, 0xcb, 0x301e5412, 0x0, 0xd5fb1121)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:4158 +0x314
go.thethings.network/lorawan-stack/pkg/ttnpb.(*GatewayConnectionStats).XXX_Marshal(0xa94f380, 0xabffad0, 0x0, 0xcb, 0xa94f300, 0xe7d33501, 0xaf1b110, 0x812373b, 0xaf1b15e, 0xade2ba0)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:1501 +0x46
github.com/gogo/protobuf/proto.Marshal(0x95fa230, 0xa94f380, 0x62, 0x62, 0x62, 0x1, 0x2)
	/home/travis/gopath/pkg/mod/github.com/gogo/protobuf@v1.3.1/proto/table_marshal.go:2940 +0x203
go.thethings.network/lorawan-stack/pkg/redis.MarshalProto(0x95fa230, 0xa94f380, 0xae254a8, 0xaf1b110, 0xae1de34, 0xaf1b110)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/redis/redis.go:49 +0x2d
go.thethings.network/lorawan-stack/pkg/redis.SetProto(0x9629a20, 0xad276c0, 0xaf1b110, 0x62, 0x95fa230, 0xa94f380, 0x0, 0x0, 0x0, 0x2, ...)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/redis/redis.go:164 +0x2d
go.thethings.network/lorawan-stack/pkg/gatewayserver/redis.(*GatewayConnectionStatsRegistry).Set(0xa96c000, 0x9601d20, 0xaed1c60, 0xade2ba0, 0x14, 0xa9da7a0, 0x0, 0xa94f380, 0x8e1dacd, 0xaed1c40)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/redis/registry.go:41 +0x1f8
go.thethings.network/lorawan-stack/pkg/gatewayserver.(*GatewayServer).UpdateConnectionStats(0xab1c190, 0x9601d20, 0xaed1c60, 0xa8c0c00, 0xae1df01, 0xae1df54)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:659 +0x71
go.thethings.network/lorawan-stack/pkg/gatewayserver.(*GatewayServer).updateConnStats(0xab1c190, 0xa8c0c00, 0xafd5580)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:678 +0x16e
created by go.thethings.network/lorawan-stack/pkg/gatewayserver.(*GatewayServer).Connect
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:451 +0xf74
```

rvolosatovs, 2020-03-17:
They just keep coming

<details>
<summary>Log</summary>

```
 ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:59384: use of closed network connection gateway_uid=eui-aaee000000000000 namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:59384 test_name=TestGatewayServer

 ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:59478: use of closed network connection gateway_uid=eui-aaee000000000000 namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:59478 test_name=TestGatewayServer

panic: runtime error: index out of range [-1]

goroutine 1683 [running]:

go.thethings.network/lorawan-stack/pkg/ttnpb.encodeVarintGateway(0xc00018e000, 0xcb, 0xcb, 0x0, 0xc, 0xcb)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:4225 +0xa9

go.thethings.network/lorawan-stack/pkg/ttnpb.(*GatewayConnectionStats).MarshalToSizedBuffer(0xc0007228a0, 0xc00018e000, 0xcb, 0xcb, 0x21, 0xc0000c41e0, 0xc0008a69b0)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:4158 +0x2f7

go.thethings.network/lorawan-stack/pkg/ttnpb.(*GatewayConnectionStats).XXX_Marshal(0xc0007228a0, 0xc00018e000, 0x0, 0xcb, 0xc000722800, 0xffffff01, 0x0, 0x1, 0x1a01e01, 0x7)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:1501 +0x5b

github.com/gogo/protobuf/proto.Marshal(0x1e083c0, 0xc0007228a0, 0xc0008a6ad8, 0xc000937460, 0xc000937460, 0x1921d60, 0xc0008a6af8)

	/home/travis/gopath/pkg/mod/github.com/gogo/protobuf@v1.3.1/proto/table_marshal.go:2940 +0x2d6

go.thethings.network/lorawan-stack/pkg/redis.MarshalProto(0x1e083c0, 0xc0007228a0, 0xc000136930, 0xc0008a6c98, 0xc000136930, 0x62)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/redis/redis.go:49 +0x50

go.thethings.network/lorawan-stack/pkg/redis.SetProto(0x1e5be40, 0xc0005e1aa0, 0xc000136930, 0x62, 0x1e083c0, 0xc0007228a0, 0x0, 0x2, 0xc000615630, 0x0)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/redis/redis.go:167 +0x39

go.thethings.network/lorawan-stack/pkg/gatewayserver/redis.(*GatewayConnectionStatsRegistry).Set(0xc0003fca68, 0x1e18ac0, 0xc0008d5020, 0xc0004f3160, 0x14, 0xc0008be498, 0x0, 0xc0007228a0, 0x146cfac, 0xc0008d1dc0)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/redis/registry.go:41 +0x24f

go.thethings.network/lorawan-stack/pkg/gatewayserver.(*GatewayServer).UpdateConnectionStats(0xc000427b90, 0x1e18ac0, 0xc0008d5020, 0xc0003e8270, 0x44f501, 0xc000a00148)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:659 +0xa6

go.thethings.network/lorawan-stack/pkg/gatewayserver.(*GatewayServer).updateConnStats(0xc000427b90, 0xc0003e8270, 0xc00094efc0)

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:678 +0x239

created by go.thethings.network/lorawan-stack/pkg/gatewayserver.(*GatewayServer).Connect

	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:451 +0x12be

FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver	22.446s
```

</details>

johanstokking, 2020-03-17:
@neoaggelos @KrishnaIyer can you coordinate who looks into this? We might have different things to fix.

neoaggelos, 2020-03-17:
There have been changes in #2108 that make https://github.com/TheThingsNetwork/lorawan-stack/issues/1332#issuecomment-600045502 and https://github.com/TheThingsNetwork/lorawan-stack/issues/1332#issuecomment-598169195 irrelevant.

Also, even though it is not a real fix for the problem, adding a bigger timeout here https://github.com/TheThingsNetwork/lorawan-stack/pull/2108/files?file-filters%5B%5D=.go#diff-b91e68ddfed2ddb93b15d4b9ab05bde0R1045 seems to have cleaned up a lot of the flakiness.

In general, everything related with statistics are being looked at as part of #2108.

neoaggelos, 2020-03-18:
Upon loads of testing, all of the tests above (and the ones that keep breaking the CI a few times a day) can be grouped in (so far) 4 categories:

1. `concurrent write to websocket connection`: AFAIK this has been fixed already in #2153 

2. `Expected deadline exceeded with valid auth, but have error:unknown:unknown (context deadline exceeded)` This happens in the `Authentication` tests (e.g.https://github.com/TheThingsNetwork/lorawan-stack/blob/3e7ae2ca9f1180cf3d937acf11bf1637bc29ca80/pkg/gatewayserver/gatewayserver_test.go#L559-L562. The error is wrong not because the test has failed, but because the context expires before the authentication can be even validated. This is 100% reproduced if we call `cancel()` before `ptc.Link`. Suggested fix would be to add a small delay before cancelling the context in line 562.

3. `Expected: 'N+1'  'Actual':   'N'` These happen because the timeout is not enough for the frontend to actually handle and account for the message. The mechanism that updates the stats is not available externally, so the best we can do is wait long enough. Naively increasing the test timeout is not a proper solution. Suggested fix would be to retry the check until it succeeds, capping the maximum allowed number of retries.

4.   `panic: runtime error: index out of range [-1]` This appears irregularly when storing the gateway connection stats on Redis, and it happens while marshalling the proto. The code has changed in #2108, so the tests are being looked at there. Looks like a race condition but I have not been able to identify the source.

rvolosatovs, 2020-03-18:
FYI:
2. We have `test.Delay` for that
3. See https://github.com/TheThingsNetwork/lorawan-stack/blob/da4c2ac4044188c6c28fc6d799d20bbb1918b9fb/pkg/networkserver/networkserver_flow_test.go#L621-L634 for an example of this approach (using `test.Delay` as well)

neoaggelos, 2020-03-18:
A new one:

```
--- FAIL: TestTraffic (0.03s)
    mqtt_test.go:169: Connection timeout
FAIL
FAIL    go.thethings.network/lorawan-stack/pkg/gatewayserver/io/mqtt    0.104s
```

johanstokking, 2020-03-19:
> 2\. Suggested fix would be to add a small delay before cancelling the context in line 562.

Sounds good

> 3\. Suggested fix would be to retry the check until it succeeds, capping the maximum allowed number of retries.

Sounds good too

> 4\. `panic: runtime error: index out of range [-1]` This appears irregularly when storing the gateway connection stats on Redis, and it happens while marshalling the proto. The code has changed in [...]

This one can be serious. This may happen when you try to marshal a `nil` struct or when the struct changes while it is being marshaled. Those are both implementation errors.

neoaggelos, 2020-03-27:
ERROR Request error                            duration=259.278µs error=error:pkg/gatewayserver/io/basicstationlns:empty_gateway_eui (empty gateway EUI) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:36438 request_id=01E4CWTRYGCFG9R79E322774MJ status=500 test_name=TestDiscover url=/router-info
2020/03/27 04:23:43 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020/03/27 04:23:43 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
 ERROR Request error                            duration=136.484µs error=error:pkg/gatewayserver/io/basicstationlns:empty_gateway_eui (empty gateway EUI) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:36440 request_id=01E4CWTRYGVGM6XQ4KMR66T3B7 status=500 test_name=TestDiscover url=/router-info
2020/03/27 04:23:43 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020/03/27 04:23:43 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
 ERROR Request error                            duration=1.681437ms error=error:pkg/gatewayserver/io/basicstationlns:empty_gateway_eui (empty gateway EUI) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:36442 request_id=01E4CWTRYHK6KM1CCM7G5TYVZM status=500 test_name=TestDiscover url=/router-info
2020/03/27 04:23:43 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020/03/27 04:23:43 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020/03/27 04:23:43 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020/03/27 04:23:43 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020/03/27 04:23:43 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020/03/27 04:23:43 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020/03/27 04:23:43 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020/03/27 04:23:43 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
--- FAIL: TestRTT (0.04s)
    --- FAIL: TestRTT/RepeatedTxAck (0.00s)
        doc.go:66: 
            /home/neo/Documents/ttn/network/pkg/gatewayserver/io/basicstationlns/basicstationlns_test.go:1233
            Expected: '2'
            Actual:   '1'
            (Should be equal)
        basicstationlns_test.go:1234: Incorrect Stats entries recorded: 1
FAIL
FAIL    go.thethings.network/lorawan-stack/pkg/gatewayserver/io/basicstationlns 0.247s

johanstokking, 2020-04-03:
@neoaggelos what is the status here?

neoaggelos, 2020-04-03:
With #2260 and #2190 merged, we should be in a much better place. I wouldn't close it yet, let's give Travis a few days to see how it fares with the changes.

neoaggelos, 2020-04-05:
New one:

```
--- FAIL: TestFlow (0.00s)
    doc.go:66: 
        /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/io/io_test.go:111
        Expected: true
        Actual:   false
    doc.go:66: 
        /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/io/io_test.go:112
        Expected 'nil' to NOT be nil (but it was)!
    doc.go:66: 
        /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/io/io_test.go:113
        Expected '2562047h47m16.854775807s' to be less than '64ms' (but it wasn't)!
FAIL
FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver/io	0.020s
```

neoaggelos, 2020-04-05:
Also, got this one again:

```
 ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:45706: use of closed network connection gateway_uid=eui-aaee000000000000 namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:45706 test_name=TestGatewayServer
 ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:45912: use of closed network connection gateway_uid=eui-aaee000000000000 namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:45912 test_name=TestGatewayServer
panic: runtime error: index out of range [-1]

goroutine 1721 [running]:
go.thethings.network/lorawan-stack/pkg/ttnpb.encodeVarintGateway(0xc0000aeb60, 0xcb, 0xcb, 0x0, 0xc, 0xcb)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/ttnpb/gateway.pb.go:4225 +0xa9
go.thethings.network/lorawan-stack/pkg/ttnpb.(*GatewayConnectionStats).MarshalToSizedBuffer(0xc000a9e4e0, 0xc0000aeb60, 0xcb, 0xcb, 0x21, 0xc000a9e360, 0xc0004ed6f0)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/ttnpb/gateway.pb.go:4158 +0x2f4
go.thethings.network/lorawan-stack/pkg/ttnpb.(*GatewayConnectionStats).XXX_Marshal(0xc000a9e4e0, 0xc0000aeb60, 0x0, 0xcb, 0xc000a9e400, 0x1, 0x1d6dbc0, 0xc000897710, 0x1d8b120, 0x2acf700)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/ttnpb/gateway.pb.go:1501 +0x5b
github.com/gogo/protobuf/proto.Marshal(0x1d611c0, 0xc000a9e4e0, 0x18fa180, 0xc000a9e360, 0xc000881220, 0xc000881340, 0xc00089e1e8)
	/home/runner/go/pkg/mod/github.com/gogo/protobuf@v1.3.1/proto/table_marshal.go:2940 +0x2d6
go.thethings.network/lorawan-stack/pkg/redis.MarshalProto(0x1d611c0, 0xc000a9e4e0, 0x40c245, 0x1934f20, 0x19396e0, 0x0)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/redis/redis.go:49 +0x50
go.thethings.network/lorawan-stack/pkg/redis.SetProto(0x7f731299a868, 0xc0008854c0, 0xc0007ba4d0, 0x69, 0x1d611c0, 0xc000a9e4e0, 0x0, 0x3, 0xc0004edaa8, 0x12bdd43)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/redis/redis.go:167 +0x39
go.thethings.network/lorawan-stack/pkg/gatewayserver/redis.(*GatewayConnectionStatsRegistry).Set.func1(0x1da3f80, 0xc0008854c0, 0x7f733c4c07d0, 0x0)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/gatewayserver/redis/registry.go:64 +0x693
github.com/go-redis/redis.(*Pipeline).pipelined(0xc0008854c0, 0xc000897c20, 0x13b77f7, 0x17668e0, 0xc000a932f0, 0x0, 0xc0008854c0)
	/home/runner/go/pkg/mod/github.com/go-redis/redis@v6.15.7+incompatible/pipeline.go:111 +0x40
github.com/go-redis/redis.(*Pipeline).Pipelined(0xc0008854c0, 0xc000897c20, 0xc0008854c0, 0x30, 0x185a180, 0x1, 0xc000897c20)
	/home/runner/go/pkg/mod/github.com/go-redis/redis@v6.15.7+incompatible/pipeline.go:120 +0x35
```

johanstokking, 2020-04-06:
That last one seems like the deepcopy didn't work out?

Isn't this a specific case in testing?

neoaggelos, 2020-04-06:
And another one:

```
 ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:39250: use of closed network connection gateway_uid=eui-aaee000000000000 namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:39250 test_name=TestGatewayServer
--- FAIL: TestGatewayServer (36.35s)
    --- FAIL: TestGatewayServer/DetectDisconnect/grpc#01 (0.77s)
        gatewayserver_test.go:650: Expected deadline exceeded on second connection but have error:unknown:unknown (context deadline exceeded)
FAIL
coverage: 75.7% of statements
FAIL	go.thethings.network/lorawan-stack/pkg/gatewayserver	36.387s
```

neoaggelos, 2020-04-06:
> That last one seems like the deepcopy didn't work out?

Yeah, probably the gateway status is not the only thing. But I really cannot find what else could be causing the issue here. Could it be that somehow the tests are led in an illegal state (e.g. same stats being updated by two sources at the same time, gc taking over and releasing stuff that it shouldn't)?

johanstokking, 2020-04-07:
> This could be solved by employing the GS events again, like we did for the `Authentication` tests, but this one is a bit more complicated.

I think this one is similar to the previous one; this is also doing a full `Link()` in the test, while the context gets already canceled before a proper setup. So you might need to double the `timeout` for the `Link()` again.

Note that GS tests already take 36 seconds, which is quite a lot. You might start experimenting lowering the general GS `timeout` value, and increasing it in smart places, like linking and potentially traffic handling.

> Yeah, probably the gateway status is not the only thing. But I really cannot find what else could be causing the issue here. Could it be that somehow the tests are led in an illegal state (e.g. same [...]

That could very well be. You can test that with `-run` (to run the particular test) and `-race` to detect races.

neoaggelos, 2020-04-11:
> Note that GS tests already take 36 seconds, which is quite a lot. You might start experimenting lowering the general GS `timeout` value, and increasing it in smart places, like linking and potentially traffic handling.

Which is why I am mostly reluctant to simply increasing timeouts when Travis fails. I think the tests would benefit a great deal if we used events instead of timeouts.  

Also, I think decreasing the timeout will only lead to more flaky tests (I don't really know how the values being used came to be, so maybe @KrishnaIyer would be more fit for this one.)

johanstokking, 2020-04-13:
I'm fine using events better, but we cannot have events for failed links if there are preconditions that are not met. It is inevitable that something happens asynchronous and there's no event for failure, while we want it to fail, i.e. we need to wait to be sure that it didn't succeed.

johanstokking, 2020-05-14:
@neoaggelos we have too many of these:

```
 ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:56960: use of closed network connection gateway_uid=eui-aaee000000000000 namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:56960 test_name=TestGatewayServer
panic: runtime error: index out of range [-1]
goroutine 780 [running]:
go.thethings.network/lorawan-stack/v3/pkg/ttnpb.encodeVarintGateway(0xc1e6700, 0xdd, 0xdd, 0x0, 0xc, 0x0, 0xdd)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:4410 +0x1b6
go.thethings.network/lorawan-stack/v3/pkg/ttnpb.(*GatewayConnectionStats).MarshalToSizedBuffer(0xbe2d580, 0xc1e6700, 0xdd, 0xdd, 0x1d4e291c, 0x0, 0xd64ead82)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:4298 +0x4b6
go.thethings.network/lorawan-stack/v3/pkg/ttnpb.(*GatewayConnectionStats).XXX_Marshal(0xbe2d580, 0xc1e6700, 0x0, 0xdd, 0xbe2d500, 0x909ab01, 0x42, 0xbfa92d8, 0xbc536d0, 0x906ef60)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/ttnpb/gateway.pb.go:1503 +0x46
github.com/gogo/protobuf/proto.Marshal(0x95e0130, 0xbe2d580, 0x0, 0x4, 0x55555555, 0x7, 0x10)
	/home/travis/gopath/pkg/mod/github.com/gogo/protobuf@v1.3.1/proto/table_marshal.go:2940 +0x203
go.thethings.network/lorawan-stack/v3/pkg/redis.MarshalProto(0x95e0130, 0xbe2d580, 0xc1ceaa8, 0x8050b39, 0x926a420, 0x926eb60)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/redis/redis.go:49 +0x3d
go.thethings.network/lorawan-stack/v3/pkg/redis.SetProto(0xe7ce4a88, 0xc2d88d0, 0xbdefb20, 0x69, 0x95e0130, 0xbe2d580, 0x0, 0x0, 0xffffffe1, 0x0, ...)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/redis/redis.go:167 +0x2d
go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/redis.(*GatewayConnectionStatsRegistry).Set.func1(0x9607320, 0xc2d88d0, 0xf7f773ec, 0x20)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/redis/registry.go:64 +0x597
github.com/go-redis/redis.(*Pipeline).pipelined(0xc2d88d0, 0xc1cea40, 0x8dbb048, 0x90bc700, 0xbfa9510, 0xc0b4f60, 0xc2d88d0)
	/home/travis/gopath/pkg/mod/github.com/go-redis/redis@v6.15.7+incompatible/pipeline.go:111 +0x32
github.com/go-redis/redis.(*Pipeline).Pipelined(0xc2d88d0, 0xc1cea40, 0xc2d88d0, 0x20, 0x91763a0, 0x1, 0xc1cea40)
	/home/travis/gopath/pkg/mod/github.com/go-redis/redis@v6.15.7+incompatible/pipeline.go:120 +0x29
github.com/go-redis/redis.(*Client).Pipelined(0xbdbb260, 0xc1cea40, 0x92a3af7, 0x1c, 0xa1e9ca0, 0x14, 0x1d4e291c)
	/home/travis/gopath/pkg/mod/github.com/go-redis/redis@v6.15.7+incompatible/redis.go:457 +0x3c
go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/redis.(*GatewayConnectionStatsRegistry).Set(0xbddf928, 0x95e8160, 0xc1c4520, 0xbc3ff40, 0x14, 0xbdda390, 0x0, 0xbe2d580, 0x9010000, 0x0, ...)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/redis/registry.go:51 +0x156
go.thethings.network/lorawan-stack/v3/pkg/gatewayserver.(*GatewayServer).UpdateConnectionStats(0xbd4d270, 0xbd31000, 0x9010000, 0xc24ade0, 0x0)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:883 +0x9e
go.thethings.network/lorawan-stack/v3/pkg/gatewayserver.(*GatewayServer).updateConnStats(0xbd4d270, 0xbd31000, 0xbfcba80)
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:673 +0x91
created by go.thethings.network/lorawan-stack/v3/pkg/gatewayserver.(*GatewayServer).Connect
	/home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/gatewayserver.go:458 +0xff1
```

Please resolve

johanstokking, 2020-05-19:
New one;

```
 ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:43472: use of closed network connection gateway_uid=eui-aaee000000000000@foo-tenant namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:43472 tenant_id=foo-tenant test_name=TestGatewayServer
--- FAIL: TestGatewayServer (36.36s)
    --- FAIL: TestGatewayServer/DetectDisconnect/grpc#01 (0.77s)
        gatewayserver_test.go:653: Expected deadline exceeded on second connection but have error:unknown:unknown (context deadline exceeded)
FAIL
FAIL	go.thethings.network/lorawan-stack/v3/pkg/gatewayserver	36.408s
```

This needs finetuning of the timeout

kschiffer, 2020-05-28:
```
ERROR Send failed, close session               error=read tcp 127.0.0.1:1882->127.0.0.1:41304: use of closed network connection gateway_uid=eui-aaee000000000000 namespace=gatewayserver/io/mqtt protocol=mqtt remote_addr=127.0.0.1:41304 test_name=TestGatewayServer
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x40 pc=0x144bd95]

goroutine 90 [running]:
go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io.(*Connection).Context(...)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/gatewayserver/io/io.go:162
go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io/udp.(*srv).gc.func1(0x18ff0a0, 0xc00097dcb8, 0x1679ba0, 0xc0005ac000, 0x1)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/gatewayserver/io/udp/udp.go:425 +0xf5
sync.(*Map).Range(0xc0001aaf98, 0xc0004a3f58)
	/opt/hostedtoolcache/go/1.14.3/x64/src/sync/map.go:333 +0x1c5
go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io/udp.(*srv).gc(0xc0001aaf20)
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/gatewayserver/io/udp/udp.go:421 +0xdf
created by go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io/udp.Serve
	/home/runner/work/lorawan-stack/lorawan-stack/pkg/gatewayserver/io/udp/udp.go:72 +0x1e2
```

htdvisser, 2020-06-09:
#2624 fixed that last comment.

bafonins, 2020-07-01:
```
--- FAIL: TestSubBandEIRPOverride (0.00s)
    doc.go:66: 
        /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/io/io_test.go:476
        Expected: true
        Actual:   false
    doc.go:66: 
        /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/io/io_test.go:477
        Expected: '1'
        Actual:   '0'
        (Should be equal)
    doc.go:66: 
        /home/travis/gopath/src/go.thethings.network/lorawan-stack/pkg/gatewayserver/io/io_test.go:478
        Expected '2562047h47m16.854775807s' to be less than '64ms' (but it wasn't)!
FAIL
FAIL	go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io	0.087s
```

rvolosatovs, 2020-07-01:
This looks like an int overflow, which is quite scary. Please ensure that this is something caused by obscurities of the test and not something that can happen in production.

rvolosatovs, 2020-07-13:
Another one:
```
2020-07-13T11:28:13.8444055Z  ERROR Request error                            duration=218.2µs error=error:pkg/gatewayserver/io/basicstationlns:empty_gateway_eui (empty gateway EUI) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:35116 request_id=01ED3YVKE5E4MZCJ6TJDV5KMH6 status=500 test_name=TestDiscover url=/router-info
2020-07-13T11:28:13.8444793Z 2020/07/13 11:27:13 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020-07-13T11:28:13.8445194Z 2020/07/13 11:27:13 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020-07-13T11:28:13.8446392Z  ERROR Request error                            duration=138.4µs error=error:pkg/gatewayserver/io/basicstationlns:empty_gateway_eui (empty gateway EUI) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:35118 request_id=01ED3YVKE5Z7A5CRAJWN5ARF1P status=500 test_name=TestDiscover url=/router-info
2020-07-13T11:28:13.8447014Z 2020/07/13 11:27:13 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020-07-13T11:28:13.8447418Z 2020/07/13 11:27:13 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020-07-13T11:28:13.8449101Z  ERROR Request error                            duration=137.8µs error=error:pkg/gatewayserver/io/basicstationlns:empty_gateway_eui (empty gateway EUI) method=GET namespace=gatewayserver/io/basicstation remote_addr=127.0.0.1:35120 request_id=01ED3YVKE6SKB9TPP9BN74YD74 status=500 test_name=TestDiscover url=/router-info
2020-07-13T11:28:13.8449873Z 2020/07/13 11:27:13 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020-07-13T11:28:13.8453477Z 2020/07/13 11:27:13 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020-07-13T11:28:13.8454049Z 2020/07/13 11:27:13 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020-07-13T11:28:13.8454795Z 2020/07/13 11:27:13 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020-07-13T11:28:13.8455937Z 2020/07/13 11:27:13 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020-07-13T11:28:13.8456759Z 2020/07/13 11:27:13 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020-07-13T11:28:13.8457157Z 2020/07/13 11:27:13 http: response.WriteHeader on hijacked connection from github.com/labstack/echo/v4.(*Response).WriteHeader (response.go:63)
2020-07-13T11:28:13.8459526Z 2020/07/13 11:27:13 http: response.Write on hijacked connection from github.com/labstack/echo/v4.(*Response).Write (response.go:72)
2020-07-13T11:28:13.8460320Z --- FAIL: TestRTT (0.15s)
2020-07-13T11:28:13.8461043Z     --- FAIL: TestRTT/FollowUpTxAck (0.13s)
2020-07-13T11:28:13.8461263Z         doc.go:66: 
2020-07-13T11:28:13.8461807Z             /home/runner/work/lorawan-stack/lorawan-stack/pkg/gatewayserver/io/basicstationlns/basicstationlns_test.go:1235
2020-07-13T11:28:13.8462139Z             Expected: '1'
2020-07-13T11:28:13.8462441Z             Actual:   '0'
2020-07-13T11:28:13.8462601Z             (Should be equal)
2020-07-13T11:28:13.8462766Z         basicstationlns_test.go:1236: Incorrect Stats entries recorded: 0
2020-07-13T11:28:13.8462920Z FAIL
2020-07-13T11:28:13.8463066Z coverage: 70.2% of statements
2020-07-13T11:28:13.8463424Z FAIL	go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io/basicstationlns	0.545s
```

neoaggelos, 2020-08-21:
Last comment is very old, closing. Re-open as needed.

closed 2020-08-21 by neoaggelos