TheThingsNetwork/lorawan-stack #1350: ttn-lw-cli alias not working on Ubuntu

loganmc10, 2019-09-11:
#### Summary

I am following this guide:
https://thethingsstack.io/v3.1.2/guides/getting-started/installation/

But snap doesn't seem to allow the ttn-lw-cli alias, I'm not sure how to get access to that command

```
ubuntu@ip-172-30-0-16:~$ snap list
Name              Version    Rev   Tracking  Publisher        Notes
amazon-ssm-agent  2.3.662.0  1455  stable/…  aws✓             classic
core              16-2.41    7713  stable    canonical✓       core
ttn-lw-stack      3.1.2      37    stable    rvolosatovs-ttn  -

ubuntu@ip-172-30-0-16:~$ sudo snap alias ttn-lw-stack.ttn-lw-cli ttn-lw-cli
error: cannot perform the following tasks:
- Setup manual alias "ttn-lw-cli" => "ttn-lw-cli" for snap "ttn-lw-stack" (cannot enable alias "ttn-lw-cli" for "ttn-lw-stack", target application "ttn-lw-cli" does not exist)
```

adriansmares, 2019-09-13:
I can confirm the report on my machine.
```
adriansmares@A98BCD2222F1:~/go/src/go.thethings.network/lorawan-stack$ sudo snap version
snap    2.41
snapd   2.41
series  16
ubuntu  19.04
kernel  5.0.0-27-generic
```

```
ame:      ttn-lw-stack
summary:   Open source LoRaWAN Network Server
publisher: The Things Network (rvolosatovs-ttn)
contact:   stack@thethingsnetwork.org
license:   unset
description: |
....
commands:
  - ttn-lw-stack
snap-id:      Qt4j7Cz4cqDwDZ8OHEvh1N3gEN9VzfSf
tracking:     stable
refresh-date: today at 11:20 CEST
channels:
  stable:    3.1.2 2019-09-05 (37) 13MB -
  candidate: ↑                          
  beta:      ↑                          
  edge:      ↑                          
installed:   3.1.2      
```

The `ttn-lw-cli` command is not exported. I think that at https://github.com/TheThingsNetwork/lorawan-stack/blob/6463643737d52a97e665a6d6107f85f4dd6fa269/.goreleaser.yml#L106-L120
we should add either `builds: cli` as well or create a separate package for the CLI itself. Thoughts on this @rvolosatovs ?

rvolosatovs, 2019-09-13:
https://github.com/TheThingsNetwork/lorawan-stack/pull/1356
I had an impression that `builds` refer to Archive IDs, but that turns out to be not true

PR #1356 (2019-09-13): Fix missing CLI in NFPM and Snapcraft

#### Summary

Closes #1350 

#### Changes

- Include CLI in NFPM and Snapcraft

#### Release Notes

- Fixed missing CLI in `deb`, `rpm` and Snapcraft packages
closed 2019-09-13 by rvolosatovs