TheThingsNetwork/lorawan-stack #1359: Github Actions

htdvisser, 2019-09-13:
#### Summary

This issue tracks progress on setting up workflows for Github Actions.

#### Why do we need this?

It would be nice if we can rely less on Travis.

#### What is already there? What do you see now? What is missing? What do you want to see?

So far we've done CI of this repository using Travis, but it's been giving us headaches lately. We got into the Github Actions beta program today, so we can start playing around. I created the [feature/actions](https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/actions) for that.

htdvisser, 2019-09-13:
https://github.com/TheThingsNetwork/lorawan-stack/runs/221176502 is the first run of the workflow. Unfortunately the Redis service isn't working properly:

> MISCONF Redis is configured to save RDB snapshots, but it is currently not able to persist on disk. Commands that may modify the data set are disabled, because this instance is configured to report errors during writes if RDB snapshotting fails (stop-writes-on-bgsave-error option). Please check the Redis logs for details about the RDB error.

Refs https://github.com/actions/example-services/issues/7

htdvisser, 2020-07-14:
@rvolosatovs please check what's still TODO and close this issue if everything is ready.

rvolosatovs, 2020-07-14:
We don't have `.travis.yml` anymore, so this issue can be closed IMO.
We still have some unimplemented features from Travis though:
- #2797 
- gcloud test in `pkg/blob` (should file an issue)

closed 2020-07-14 by rvolosatovs