TheThingsNetwork/lorawan-stack #1362: RPC correlation IDs not propagated to IS right calls

rvolosatovs, 2019-09-17:
#### Summary

RPC correlation IDs are not propagated to IS right calls

#### Steps to Reproduce

1. Perform a `asns.LinkApplication` call
2. Observe a set `X` of correlation IDs generated for the `LinkApplication` RPC request
3. Observe a set `Y` of correlation IDs generated and passed to `ListRights` RPC on IS

(link to test line here will be added here)

#### What do you see now?

Original RPC correlation IDs lost when `ListRights` RPC is performed

#### What do you want to see instead?

Original RPC correlation IDs merged with the `ListRights` correlation IDs when RPC is performed

#### How do you propose to implement this?

Not sure

#### Can you do this yourself and submit a Pull Request?

no

htdvisser, 2019-09-18:
This isn't a bug. Correlation IDs weren't designed to do this. They are meant to correlate LoRaWAN traffic, and they aren't propagated by default on any RPC. The only way to propagate Correlation IDs is to put them in the request messages, like what we do with uplink/downlink messages.

Why would we need to do this?

rvolosatovs, 2019-09-18:
My understanding of this was that we would use correlation IDs for tracing of all RPCs, e.g. we could see the full chain of `LinkApplication` -> `ListRights` -> ???.
Closing this for now, please reopen if you think it's necessary.

closed 2019-09-18 by rvolosatovs

htdvisser, 2019-09-18:
No, for distributed tracing of HTTP calls and RPCs we use OpenCensus (which is becoming OpenTelemetry)