TheThingsNetwork/lorawan-stack #1375: Goroutine leak in GS scheduling

htdvisser, 2019-09-23:
#### Summary

There seems to be a goroutine leak in GS scheduling.

#### Steps to Reproduce

1. Start The Things Stack and let it run for a while
2. Run a pprof on goroutines

#### What do you see now?

[image]

[image]

#### What do you want to see instead?

I'd expect the number of goroutines to be more stable, and definitely not this many in GS scheduling GC.

#### Environment

e69e3e135

johanstokking, 2019-09-23:
The problem is that the cancelable context is initiated in `io.NewConnection()`, while it has before been passed to `NewScheduler()`. So, on disconnect, the context that the scheduler used is not canceled.

Change the initialization order a bit. It's not just moving code because `NewConnection()` takes the scheduler. We may have `NewConnection()` instantiate the scheduler with the cancelable context, if that doesn't create nasty dependencies.

KrishnaIyer, 2019-09-23:
@johanstokking : This is already fixed in `master` via my latest PR ([dc71f54](https://github.com/TheThingsNetwork/lorawan-stack/commit/dc71f540b328f9cb0e6ae5eba3c0586a3b8d51f8)).

KrishnaIyer, 2019-09-23:
This can be closed once tested on the demo env.

htdvisser, 2019-09-24:
This indeed seems to be resolved in 67cc2b31b

closed 2019-09-24 by htdvisser