TheThingsNetwork/lorawan-stack #1382: Fieldmasks are not respected by CLI `get` commands

rvolosatovs, 2019-09-25:
#### Summary

Some fieldmasks are not respected by CLI

#### Steps to Reproduce

1. cli device get test-app test-dev --mac-state.device-class

#### What do you see now?

```
{
  "ids": {
    "device_id": "test-dev",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "0004A30B001BC7AF",
    "join_eui": "4200000000000000",
    "dev_addr": "42FFFFFF"
  },
  "created_at": "2019-09-25T12:05:53.084Z",
  "updated_at": "2019-09-25T13:42:55.864212220Z",
  "network_server_address": "localhost",
  "mac_state": {
    "current_parameters": {

    },
    "desired_parameters": {

    }
  }
}
```

#### What do you want to see instead?

```
{
  "ids": {
    "device_id": "test-dev",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "0004A30B001BC7AF",
    "join_eui": "4200000000000000",
    "dev_addr": "42FFFFFF"
  },
  "created_at": "2019-09-25T12:05:53.084Z",
  "updated_at": "2019-09-25T13:42:55.864212220Z",
  "mac_state": {
     "device_class": <CLASS>
  }
}
```

#### How do you propose to implement this?

@htdvisser is responsible for CLI, so he should know better

#### Can you do this yourself and submit a Pull Request?

no

htdvisser, 2019-09-25:
Is your device class A by any chance? In that case, it makes sense.

From https://developers.google.com/protocol-buffers/docs/proto3:

> If a value is missing in the JSON-encoded data or if its value is null, it will be interpreted as the appropriate default value when parsed into a protocol buffer. If a field has the default value in the protocol buffer, it will be omitted in the JSON-encoded data by default to save space. An implementation may provide options to emit fields with default values in the JSON-encoded output.

rvolosatovs, 2019-09-25:
Yes, it is class A indeed - that's not the biggest issue IMO. What's up with
```
  "network_server_address": "localhost",
  "mac_state": {
    "current_parameters": {

    },
    "desired_parameters": {

    }
```
part?

htdvisser, 2019-09-25:
The CLI just renders protos using jsonpb, so this is something in our proto definitions or gogo options. I suspect this has something to do with the `nullable = false` options on `current_parameters` and `desired_parameters`.

rvolosatovs, 2019-09-25:
This is confusing to the user and has to be fixed.
This should also be very trivial to fix - just decode the JSON into a `map[string]interface{}` and loop over the fieldmasks to only include the specified ones.

htdvisser, 2019-09-26:
Sounds good, go ahead.

htdvisser, 2019-11-12:
Let's wait for https://github.com/golang/protobuf APIv2, and then build it into a forked jsonpb

johanstokking, 2019-12-06:
The problem is not in jsonpb, it is a combination of 1) the CLI using the allowed bottom level fields and 2) the generated `SetFields()` filling in an empty struct for sub-fields, even if the source field is `nil`.

1) is necessary to merge sub-fields from various sources that would otherwise overwrite. For example, `session` comes from both NS and AS and their fields need to be merged, so we do that by the allowed bottom level fields
2) seems wrong; if the source is `nil`, the destination should also be `nil`, otherwise we start filling in stuff that leads to devices that fail validation

If we change 2 (of which I'm in favor), this not only affects CLI but all places where we get entities from the device registries, and we need to explicitly add `nil`-checks and `GetField()`-style code.

rvolosatovs, 2019-12-10:
#1713 brings only partial fix - the non-nullable fields still are rendered even though not requested(see OP)

johanstokking, 2019-12-11:
> #1713 brings only partial fix - the non-nullable fields still are rendered even though not requested(see OP)

I don't think this is import to fix, pushing that to Backlog

neoaggelos, 2021-08-18:
Is this important to keep around? @htdvisser I suppose we can just close this.

htdvisser, 2021-08-19:
This will eventually be resolved when we're done migrating away from gogo proto (https://github.com/TheThingsNetwork/lorawan-stack/issues/2798), so I don't think we need to keep this open.

closed 2021-08-19 by htdvisser