TheThingsNetwork/lorawan-stack #1386: Several NS configuration settings not settable through CLI

rvolosatovs, 2019-09-25:
#### Summary

NS configuration settings not settable through CLI
Refs https://github.com/TheThingsNetwork/lorawan-stack/pull/1329#issuecomment-534576954

#### Steps to Reproduce

1. `ttn-lw-stack --help`

#### What do you see now?

```
      --ns.cooldown-window duration                                    Time window starting right after deduplication window, during which, duplicate messages are discarded (default 1s)
      --ns.deduplication-window duration                               Time window during which, duplicate messages are collected for metadata (default 200ms)
      --ns.default-mac-settings.adr-margin float                       The default margin Network Server should add in ADR requests if not configured in device's MAC settings (default 15)
      --ns.default-mac-settings.class-b-timeout duration               Deadline for a device in class B mode to respond to requests from the Network Server if not configured in device's MAC settings (default 1m0s)
      --ns.default-mac-settings.class-c-timeout duration               Deadline for a device in class C mode to respond to requests from the Network Server if not configured in device's MAC settings (default 15s)
      --ns.default-mac-settings.desired-rx1-delay int32                Desired Rx1Delay value Network Server should use if not configured in device's MAC settings (default 5)
      --ns.default-mac-settings.status-count-periodicity uint          Number of uplink messages after which a DevStatusReq MACCommand shall be sent by Network Server if not configured in device's MAC settings (default 20)
      --ns.default-mac-settings.status-time-periodicity duration       The interval after which a DevStatusReq MACCommand shall be sent by Network Server if not configured in device's MAC settings (default 1h0m0s)
      --ns.dev-addr-prefixes strings                                   Device address prefixes of this Network Server
      --ns.downlink-priorities.join-accept string                      Priority for join-accept messages (lowest, low, below_normal, normal, above_normal, high, highest) (default "highest")
      --ns.downlink-priorities.mac-commands string                     Priority for messages carrying MAC commands (lowest, low, below_normal, normal, above_normal, high, highest) (default "highest")
      --ns.downlink-priorities.max-application-downlink string         Maximum priority for application downlink messages (lowest, low, below_normal, normal, above_normal, high, highest) (default "high")
      --ns.interop.directory string                                    Retrieve the interoperability client configuration from the filesystem
      --ns.interop.url string                                          Retrieve the interoperability client configuration from a web server
      --ns.net-id string                                               NetID of this Network Server (default "000000"
```

#### What do you want to see instead?

`ns.default-mac-settings.desired-adr-ack-{limit,delay}-exponent` etc.
There may be more cases like this.

#### How do you propose to implement this?

No idea, suggestion in https://github.com/TheThingsNetwork/lorawan-stack/pull/1329#pullrequestreview-292868899 doesn't do anything

#### Can you do this yourself and submit a Pull Request?

no

neoaggelos, 2020-05-03:
Upon initial investigation, for each missing configuration variable, because of the way the `setDefaults` function has been implemented we need to do the following:

- Add a default value in https://github.com/TheThingsNetwork/lorawan-stack/blob/c5bec9273bc348f9f4dfde4195991ee6f46d0826/pkg/networkserver/config.go#L135, similar to what `DesiredRx1Delay` has.
- Add handling for the respective type in https://github.com/TheThingsNetwork/lorawan-stack/blob/929879d83313265bc61cf78e698308fbde02a473/pkg/config/config.go#L499

For example, adding in the first file:
```
		DesiredRx1Delay: func(v ttnpb.RxDelay) *ttnpb.RxDelay { return &v }(ttnpb.RX_DELAY_5),
		DesiredADRAckDelayExponent: func(v ttnpb.ADRAckDelayExponent) *ttnpb.ADRAckDelayExponent { return &v }(ttnpb.ADR_ACK_DELAY_1),
```

and in the second one:
```

			case ttnpb.RxDelay:
				m.viper.SetDefault(name, int32(val))
				flags.Int32P(name, shorthand, int32(val), description)

			case ttnpb.ADRAckDelayExponent:
				m.viper.SetDefault(name, int32(val))
				flags.Int32P(name, shorthand, int32(val), description)

```

Properly displays the option in the help text:
```
$ stack --help | grep desired
      --ns.default-mac-settings.desired-adr-ack-delay-exponent int32   Desired ADR_ACK_DELAY value Network Server should use if not configured in device's MAC settings
      --ns.default-mac-settings.desired-rx1-delay int32                Desired Rx1Delay value Network Server should use if not configured in device's MAC settings (default 5)
```

@rvolosatovs If it is OK to add defaults for every NS option, then this issue is straightforward to solve. Otherwise, we will need fiddle a bit more with `setDefaults`.

rvolosatovs, 2020-05-04:
We can't define defaults for all values, so this has to be handled otherwise.

PR #2584 (2020-07-30): Network Server configure default MAC settings

#### Summary

Closes #1386, #2859 

#### Changes

- Refactor config handling in `pkg/config` for existing types
- Add extra handling for `*ttnpb.ADRAckDelayExponent`, `*ttnpb.ADRAckLimitExponent`, `*ttnpb.AggregatedDutyCycle`, `ttnpb.RxDelay`, and `*time.Duration` values. Hooks work with `string` values, and set the value to `nil` when the string is empty.
- Add `mapstructure` hooks for these types, support both integer values (already existing) as well as string enum values.

#### Testing

- Start stack with configuration, add extra debug message [... cut]
closed 2020-07-30 by neoaggelos