TheThingsNetwork/lorawan-stack #1394: Support multiple band-plans on a single concentrator

ama9910, 2019-09-27:
#### Summary

We would like to see support for multiple channel plans on the same gateway concentrator. 

As @htdvisser has suggested in [https://github.com/TheThingsNetwork/lorawan-stack/issues/761#issuecomment-530304855](https://github.com/TheThingsNetwork/lorawan-stack/issues/761#issuecomment-530304855), the channel frequencies are the same for AS923-925 and AU915 FSB6. A single 8-channel gateway could then service could service both bands.

#### Why do we need this?

This is important in Australia for instance, where we use both AS923 and AU915. This would also be useful in any country whose ISM band supports multiple LoRaWAN band plans. It can help device manufacturers test their products for global markets without having to reconfigure gateways.

It helps keep the cost of gateway hardware down since we won't need to install dual card gateways to achieve the same thing.

I also think this is a first in the industry and helps to show that TTI continue to release innovative solutions to problems.

#### What is already there? What do you see now?

Currently, with gateway configured to AS_923_925 and device configured to AU_915_FSB_6, uplinks are being received and processed, however downlinks fail with the error 
`stack_1      |  DEBUG Failed to schedule on path               error=error:pkg/gatewayserver/io:data_rate (no data rate with index `10`) gateway_uid=eui-00800000a0001da2 grpc_method=ScheduleDownlink grpc_service=ttn.lorawan.v3.NsGs namespace=gatewayserver request_id=01DNG8WP9JA36HTT6TP33RFGYT`

#### What is missing? What do you want to see?

According to @johanstokking in [https://github.com/TheThingsNetwork/lorawan-stack/issues/761#issuecomment-534521743](https://github.com/TheThingsNetwork/lorawan-stack/issues/761#issuecomment-534521743)

> The logs indicate that GS can't find the data rate index in the gateway's band. This is because of the mismatch between end device and gateway band. That's normal.

> The reason why my suggestion to let NS pass the frequency plan as part of the transmission request does not work, is that the GS scheduler enforces duty-cycle by (sub-)band. The band is referred to by the frequency plan. Mixing bands AS926 and AU915 is no problem for duty-cycle, but this is not a general rule. For this issue not to go off-topic, please open a separate issue to see what can be done.

#### How do you propose to implement this?

Seems like changes required to the GS to support this.

#### Can you do this yourself and submit a Pull Request?

Unfortunately not. But happy to implement the build and commit the resources and hardware required to test multiple combinations of band plans and devices.

johanstokking, 2019-09-27:
Note to self; I think we should not support this if any of the band is duty-cycle limited, because then we'd need to take the lowest common denominator of sub-bands (which probably don't align either) and keep track of that separately.

So;
- GS scheduler to support multiple frequency plans. If any is duty-cycle limited, return config error
- Extend `TxRequest` with `frequency_plan_id`
- GS to figure out the data rate by the NS given frequency plan ID (falling back to the only configured one for backward compatibility), to know the time-on-air
- GS has one list of scheduled emissions for conflict detection and scheduling any time (class C)

It's not super complicated but the GS scheduler is delicate and this requires extensive unit testing.

ama9910, 2019-09-29:
> I think we should not support this if any of the band is duty-cycle limited, because then we'd need to take the lowest common denominator of sub-bands (which probably don't align either) and keep track of that separately.

That would exclude Australia since our 915-928 MHz ISM band is duty cycle limited, regardless of band plan. Wouldn't it be valid to use the lowest duty cycle limit from both band plans and track the airtime per gateway regardless of band plan, since the duty cycle limit is a function of the ISM band, not the LoRaWAN band plan?

johanstokking, 2019-12-23:
@KrishnaIyer since https://github.com/TheThingsNetwork/lorawan-stack/pull/1682 we allow for defining the EIRP and duty-cycle per sub-band in the frequency plan. Also, we have added [AS923 variants for Australia with its EIRP](https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/11/files#diff-1eaeccb9459018646b35d244175c0162R5). This means that there should not be a difference anymore between the EIRP in the frequency plans that are assigned to the gateway. That renders [my earlier comment](https://github.com/TheThingsNetwork/lorawan-stack/issues/761#issuecomment-534521743) about not being able to support the frequency plan from the NS irrelevant. The frequency plans assigned to the gateway are more about supported channels.

So, making the frequency plan part of the `TxRequest` is back on the table. This means in fact that a gateway can have AU915 FSB 6 or AS923 AU (same channels), while the device can be using either frequency plan.

For GS scheduler, and I'll do this if you assign me to the right issue, this means that it should lookup the SF and BW from the device's frequency plan (and use that for TOA calculation etc), but take the EIRP and duty-cycle settings from the gateway's frequency plans (they should be the same).

ama9910, 2020-05-18:
FYI, I've just edited my previous comment. The Australian ISM band is not duty cycle limited. I previously thought it was, but that's incorrect.

Any chance of implementing this before V3 goes live on the public network? This will be a very useful feature for Australia and potentially other countries.

johanstokking, 2020-05-18:
Thanks. Yeah I'm pulling this to Next Up.

Besides duty cycle there are other regulatory things in the band (with overrides in frequency plan), like max EIRP and dwell time. But I guess that the AU frequency plan variant using the AS923 band matches settings of the AU915 band/plan, or at least it should match. It wouldn't make sense to have different max EIRP depending on the end device's frequency plan.

ama9910, 2020-05-18:
Yes agreed. Since the limitation is on the overall band rather than the frequency plan, we can treat both plans the same. eg, The AU ISM band has a max EIRP limitation of 30dBm. This then applies to any frequency plan we use in that band.