TheThingsNetwork/lorawan-stack #1405: Manage claim authentication code

johanstokking, 2019-10-01:
#### Summary

Manage claim authentication code in Console

#### Why do we need this?

To allow users to edit the value and set to/from dates

#### What is already there? What do you see now?

Only CLI support

#### What is missing? What do you want to see?

Console support

References https://github.com/TheThingsNetwork/lorawan-stack/issues/579

#### How do you propose to implement this?

In the Join Server section, add optional settings for the claim authentication code

The entire `claim_authentication_code` field is optional. If set, `value` is mandatory. The `valid_from` and `valid_to` date/time is optional

Please show value as "password" (that you can show)

See API for validation rules

#### Can you do this yourself and submit a Pull Request?

Can review

johanstokking, 2019-11-12:
@kschiffer @pgalic96 please coordinate who can pick this up

kschiffer, 2019-11-12:
I'd like to do it. If it's not urgent, I would do it after my November issues.

bafonins, 2020-04-02:
Closed via https://github.com/TheThingsNetwork/lorawan-stack/pull/2072

closed 2020-04-02 by bafonins