TheThingsNetwork/lorawan-stack #1423: issue: Database table `accounts` does not exist

amprodes, 2019-10-06:
:rotating_light: :stop_sign: This issue tracker is not for questions :stop_sign: :rotating_light:

Hi, guys, can you help me? I'm new with this docker thing and am installing the stack and am getting this error on running docker yml file

stack_1     |   INFO Setting up Identity Server              
stack_1     | error:cmd/internal/shared:initialize_identity_server (could not initialize Identity Server)
stack_1     | --- error:pkg/identityserver/store:database_table (database table `accounts` does not exist)
stack_1     |     table=accounts

What am I doing wrong?

htdvisser, 2019-10-07:
As the template already indicates: This issue tracker is not for questions

Please raise your question on our community platforms:

- Forum: https://www.thethingsnetwork.org/forum/
- Slack: https://thethingsnetwork.slack.com (you can self join via https://account.thethingsnetwork.org)

closed 2019-10-07 by htdvisser

amalavegecko, 2019-10-07:
I got this message "Access Denied" at the forum