TheThingsNetwork/lorawan-stack #1436: Optionally require authenticated gateway links

johanstokking, 2019-10-09:
#### Summary

Optionally require authenticated links from gateways.

#### Why do we need this?

To disallow unauthenticated connections, like UDP, when the gateway owner knows that the gateway is capable of authenticating itself (i.e. using Basic Station).

#### What is already there? What do you see now?

GS keeps track of UDP source addresses which detects and disallows spoofing gateway EUIs while the real UDP-linked gateway is still connected.

#### What is missing? What do you want to see?

We should not allow any unauthenticated gateway links, i.e. UDP or Basic Station without auth, on a per-gateway setting basis.

#### How do you propose to implement this?

- Add a `require_authentication` bool to `Gateway`, store in IS
- Get this information when connecting the gateway in GS
- Check if the concerning frontend is authenticated. If not, and if `require_authentication` is set, stop the connecting procedure

#### Can you do this yourself and submit a Pull Request?

Can review

KrishnaIyer, 2020-07-27:
Basic Station is now covered via #2948, updated description to match that.

htdvisser, 2021-03-01:
I'll do the Identity Server part and then hand over to @KrishnaIyer / @adriansmares.

PR #3911 (2021-03-24): Add field to require secure gateway connections

#### Summary

This adds a field to gateways that allows users to indicate that their gateways are only to connect over authenticated connections.

Closes #1436

#### Changes

- [x] Added API field and DB column
- [x] Updated GS connection logic to check that field
- [x] Add checkbox to Console
- [x] Updated console e2e tests

Before:
[image]

After:
[image]

#### Testing

- Manually tested
- Updated e2e tests

##### Regressions

...

#### Notes for Reviewers
closed 2021-03-24 by htdvisser