TheThingsNetwork/lorawan-stack #1439: Application upstream message `received_at` never set

johanstokking, 2019-10-09:
#### Summary

AS never sets `received_at` on the `ApplicationUp` message.

#### Steps to Reproduce

1. Subscribe to AS messages, i.e. MQTT or webhooks
2. See any upstream message, for example uplink

#### What do you see now?

No `received_at` set

#### What do you want to see instead?

For uplink-triggered messages (uplink, join-accept, downlink ack/nack), the `received_at` set to the `UplinkMessage` received from NS. In other scenarios (downlink queued/sent/failed, location solved, package-triggered messages, etc), don't set the value.

#### Can you do this yourself and submit a Pull Request?

Can review

mcoracin, 2019-10-18:
Hello,
When can we expect to have a fix for that ?
Thanks and regards,
Michael

htdvisser, 2019-10-18:
Hi @mcoracin, 

This issue is in the **Next Up** milestone, which means that we will likely plan it for next month's milestone. If by the end of this month we still have time left, we may still address it in October.

If this is very important to you, you're of course welcome to contribute a fix yourself.

rvolosatovs, 2019-11-18:
This is not what should happen IMO
`ApplicationUp.ReceivedAt` is the timestamp when application uplink is received by AS. Note, that `ApplicationUp` may be triggered by NS downlink as well, for example, in which case there's no `received_at` at the first place.
What should happen instead, if we need to know the time when uplink(i.e. data uplink or join-request) was received by the stack, we should add such field to the payload of respective uplinks - at this time I think the only one, for which it makes sense is `ApplicationUplink`.

rvolosatovs, 2019-11-18:
I suggest the following approach: https://github.com/TheThingsNetwork/lorawan-stack/compare/3d04ed8ec6e25bbb4c4b5a0f9385f7eb7e3967c6...rvolosatovs:feature/application-up-recv-at

PR #1605 (2019-11-21): Provide received_at in appropriate uplink messages

#### Summary

Closes #1439

#### Changes

- Pass the `received_at` from the `UplinkMessage` to the `ApplicationUp` that is sent to the ApplicationServer.

#### Notes for Reviewers

The initial issue mentioned that the AS should fill in the timestamp, but it was actually the NS which didn't fill it.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `REA [... cut]
closed 2019-11-21 by adriansmares