TheThingsNetwork/lorawan-stack #1455: Docker image won't start with 3.2.1 (can't find cert.pem)

loganmc10, 2019-10-11:
```
stack_1      |   INFO Setting up core component
stack_1      |   INFO Setting up Identity Server
stack_1      |   INFO Setting up Gateway Server
stack_1      | error:cmd/internal/shared:initialize_gateway_server (could not initialize Gateway Server)
stack_1      | --- error:pkg/gatewayserver:listen_frontend (failed to start frontend listener `MQTT/tls` on address `:8882`)
stack_1      |     protocol=MQTT/tls
stack_1      |     address=:8882
stack_1      | --- open cert.pem: no such file or directory
```

This didn't happen in 3.2.0. I'm using acme/lets encrypt for the cert, not self-signed

KrishnaIyer, 2019-10-11:
@loganmc10: Please fill out the issue template. While doing so, please dump your stack config.

htdvisser, 2019-10-11:
@loganmc10 Can you try with `--tls.source acme` (or corresponding env / yml config)?

Suspect: https://github.com/TheThingsNetwork/lorawan-stack/pull/1451/files?file-filters%5B%5D=.go#diff-b05b21ce72e38c01aa5a7b5d0d038661R70-R79

cc: @johanstokking

htdvisser, 2019-10-11:
We also forgot to update the documentation for these configs in #1451

- `docker-compose.yml`
- `doc/content/guides/getting-started/configuration.md`

and it looks like the release notes from the PR weren't included in the changelog either, opened #1456 and #1458 for that.

PR #1463 (2019-10-14): Fix ACME config

#### Summary

Closes #1455 

#### Changes

- Revert original behavior to prioritize enabled ACME configuration
- Deprecate ACME `enable` setting
- Update documentation
- Update change log
closed 2019-10-14 by johanstokking