TheThingsNetwork/lorawan-stack #1459: Bulk create does not respect field mask + minor issues

johanstokking, 2019-10-14:
#### Summary

Bulk create does not respect field mask. Other minor issues related to generating a claim authentication code and styling.

#### Steps to Reproduce

1. Start stack with IS + JS + DTC + Console
2. Import devices from ATECC608A-TNGLORA manifest

#### What do you see now?

Import works, good experience with progress bar etc.

There are a few issues though;

1. Base64 padding is not respected in device's certificate
2. Bulk Creation button is put behind Add Device
3. Claim authentication code is not generated as option
4. Description of format is in the dropdown

#### How do you propose to implement this?

1. Respect base64 padding in device's certificate – @johanstokking 
2. Create two buttons next to each other: **Add Single Device** and **Import Devices** – @kschiffer 
3. Add checkbox to generate claim authentication code (see below) – @kschiffer 
4. Put description under the dropdown – @kschiffer 

For 3, see #1404:

>This is basically creating all the devices. Depending on Console cluster config, this has to go to IS, JS, NS and AS. When a create fails, rollback that device (like we do now), and stop.
>
>Note that flag here; if this checkbox is checked (for all devices), generate a random 4-byte hex string (per device) and put that in `claim_authentication_code.value` (and add that to the field mask).

#### Can you do this yourself and submit a Pull Request?

Can review

kschiffer, 2019-10-14:
> Field mask is not respected; i.e. if converter returns provisioner_id as part of the template, the JS is not hit for create

That is due to the template not setting `supports_join` when join server fields are set.

johanstokking, 2019-10-15:
> That is due to the template not setting `supports_join` when join server fields are set.

This is now addressed in #1472.

Added a sixth point; the JS address does not seem to be set. I'm testing with `supports_join` is true.

PR #1476 (2019-10-16): Fix bulk create issues
#### Summary
Closes #1459
References #1086 

[image]

#### Changes
- Update device entity map (`claim_authentication_code` was missing)
- Move import button to the table view (and adapt routing accordingly)
- Rename bulk creation to device import
- Add a `FieldInfo` component, for info sections within forms
- Move display of template format description to below the select input
- Make bytestring generator generate uppercase strings due to backend validation)
- Add option to generate claim auth codes automatically during device import
- Add component hosts automatically d [... cut]
closed 2019-10-16 by kschiffer