TheThingsNetwork/lorawan-stack #1493: Allow read-only field masks for update operations if the value does not change

kschiffer, 2019-10-19:
#### Summary
When querying the RPCs through the http bridge, using read-only field masks in write operations will error out, even when the value remains the same. It would be better to ignore these field masks then rather than returning a validation error.
The console would benefit from such behavior, as we sometimes want to *upsert* entities, using the same request. This is currently not possible, as the `ids` field mask cannot be used when the entry already exists, so we need to use two different requests to achieve upserting.

#### Steps to Reproduce

Try running the following cURL (given that the `test-app` application exists and the `test-device` does not)
```shell
curl -X PUT \
  http://localhost:1885/api/v3/js/applications/test-app/devices/test-device \
  -H 'Accept: application/json, text/plain, */*' \
  -H 'Accept-Encoding: gzip, deflate' \
  -H 'Authorization: Bearer $access_token' \
  -H 'Cache-Control: no-cache' \
  -H 'Connection: keep-alive' \
  -H 'Content-Length: 784' \
  -H 'Content-Type: application/json;charset=UTF-8' \
  -H 'Host: localhost:1885' \
  -H 'cache-control: no-cache' \
  -d '{
  "end_device":{
    "root_keys":{
      "app_key":{
        "key":"665A771492D6E16FD0E271AA1D9E6A06"
      },
      "nwk_key":{
        "key":"F5B9273CBF3227041FA66ADB20569BF8"
      }
    },
    "lorawan_version":"1.0.0",
    "resets_join_nonces":false,
    "supports_class_c":false,
    "frequency_plan_id":"EU_863_870",
    "lorawan_phy_version":"1.0.0",
    "ids":{
      "device_id":"test-device",
      "application_ids":{
        "application_id":"test-app"
      },
      "join_eui":"70B3D57ED0000000",
      "dev_eui":"0004A310001FF9E2"
    },
    "supports_join":true
  },
  "field_mask":{
    "paths":[
      "root_keys.app_key.key",
      "root_keys.nwk_key.key",
      "resets_join_nonces",
      "ids.device_id",
      "ids.application_ids.application_id"
    ]
  }
}'
```
It will run through the first time, but error our the second time, because the `ids` field masks won't pass the validator. However, a `PUT` request should make no difference about that.
The requests targets the Join Server, but this holds true for all end device registries.

#### What do you see now?
Errors when using field masks of read-only props, even when the values would not change.

#### What do you want to see instead?
Such field masks being ignored without error.

#### Environment
TTS v3.2.2

#### How do you propose to implement this?
🤷‍♂, hopefully not a big deal

#### Can you do this yourself and submit a Pull Request?
@johanstokking / @rvolosatovs

rvolosatovs, 2019-10-22:
We can do this, but IMO this should not be necessary.

1. The only use case for setting `ids.device_id` and `ids.application_ids` is creation of a new device. In all other cases neither should any of these field paths be set, nor any of the values set in the proto(the IDs are inferred from the URL)
2. Creation of a device should not be an upsert - it should error if entity already exists.

Given (1.) and (2.) I don't see how this is helpful.
Can you describe your use case for this?

johanstokking, 2019-10-22:
> 1. The only use case for setting `ids.device_id` and `ids.application_ids` is creation of a new device. In all other cases neither should any of these field paths be set, nor any of the values set in the proto(the IDs are inferred from the URL)

The thing is that the client doesn't always know whether it's create or update. That is also why we don't have `Create()` or `Update()` rpcs but `Set()`. Sure, if the device exists and we're changing read-only settings, we should error, but that is unrelated to this (because we're not changing identifiers).

> 2. Creation of a device should not be an upsert - it should error if entity already exists.

It's not creation; it's setting a device. The client provides the fields it knows and doesn't care whether the device exists or not.

> Can you describe your use case for this?

The use case is importing devices on a JS-only, while there are NS and AS in the cluster. In that case, the device only gets created on IS+JS, but not on NS and AS because we don't have those fields. Now, the user wants to set the device on NS and AS, but runs into this error.

rvolosatovs, 2019-10-22:
> The use case is importing devices on a JS-only, while there are NS and AS in the cluster. In that case, the device only gets created on IS+JS, but not on NS and AS because we don't have those fields. Now, the user wants to set the device on NS and AS, but runs into this error.

This clears that up, thanks!
How high of a priority is it?

PR #1502 (2019-10-23): Allow read-only fields in update mask if they don't change

#### Summary

Closes #1493 

#### Changes

- Replace merciless `ProhibitFields()` by checking if there's a change
closed 2019-10-23 by johanstokking