TheThingsNetwork/lorawan-stack #1501: Change root key in Console fails

johanstokking, 2019-10-23:
#### Summary

Console errors when changing the device's root key.

#### Steps to Reproduce

1. Click **Add Device**
2. Fill out basic settings, not very import
3. Uncheck **External Join Server** and make sure keys get generated automatically
4. Go to created device's **General Settings**
5. Change a byte in the root key
6. Click Save

#### What do you see now?

>Invalid `end_device`: embedded message failed validation

The `POST` to the JS and NS device registry fail. This is because the primary end device identifiers are missing from this request. These should always be in the `end_device`. Current example JSON:

```json
{
  end_device: {
    root_keys: { app_key: [Object] },
    resets_join_nonces: false,
    supports_class_c: false,
    supports_join: true,
    ids: { join_eui: '70B3D57ED0000000', dev_eui: '0004A310001FF9E2' }
  },
  field_mask: { paths: [ 'root_keys.app_key.key', 'resets_join_nonces' ] }
}
```

#### What do you want to see instead?

Successful update

#### Environment

v3.2.2

#### How do you propose to implement this?

Probably add the primary identifiers to the payload

#### Can you do this yourself and submit a Pull Request?

Can review

johanstokking, 2019-10-23:
Full error:

```json
{
    "code": 3,
    "message": "error:pkg/errors:validation (invalid `end_device`: embedded message failed validation)",
    "details": [
        {
            "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
            "namespace": "pkg/errors",
            "name": "validation",
            "message_format": "invalid `{field}`: {reason}",
            "attributes": {
                "field": "end_device",
                "name": "SetEndDeviceRequestValidationError",
                "reason": "embedded message failed validation"
            },
            "correlation_id": "3139d7d2be1c4b67a47ff96e8d60f41b",
            "cause": {
                "namespace": "pkg/errors",
                "name": "validation",
                "message_format": "invalid `{field}`: {reason}",
                "attributes": {
                    "field": "ids",
                    "name": "EndDeviceValidationError",
                    "reason": "embedded message failed validation"
                },
                "cause": {
                    "namespace": "pkg/errors",
                    "name": "validation",
                    "message_format": "invalid `{field}`: {reason}",
                    "attributes": {
                        "field": "device_id",
                        "name": "EndDeviceIdentifiersValidationError",
                        "reason": "value does not match regex pattern \"^[a-z0-9](?:[-]?[a-z0-9]){2,}$\""
                    },
                    "code": 3
                },
                "code": 3
            },
            "code": 3
        }
    ]
}
```

bafonins, 2019-10-23:
This issue is not only related to `root_keys`, but to all fields updated in the general settings page. Most probably caused by [this commit](https://github.com/TheThingsNetwork/lorawan-stack/commit/16033cb795bf991a1e49ef5ac888ff5fdef68373#diff-4fa7283b7d65ffba5f75fae2b1c0d93b)