TheThingsNetwork/lorawan-stack #1506: Add Device Image to Entity Registry

mjamescompton, 2019-10-23:
#### Summary

Add an endpoint for uploading an image for a device and store in Entity Registry similar to the user profile picture.

#### Why do we need this?

Device claiming frontend requires an image to be uploaded after claiming.

#### What is already there? What do you see now?

Nothing

#### Can you do this yourself and submit a Pull Request?

no

bafonins, 2019-11-11:
@htdvisser Could you provide more specific requirements about this feature?

bafonins, 2019-11-11:
@mjamescompton is there any reason you need a separate endpoint for this? can we just extend the update rpc for end devices to include the device picture?

htdvisser, 2019-11-11:
Let's not add a separate RPC for this. Better to do it the same way as we do with user profile pictures.

closed 2019-11-25 by bafonins